robotology/robotology-superbuild #1: Introducing the robotology-superbuild

traversaro, 2017-11-24:
Hi everyone, 
    we had to cleanup the good old [`codyco-superbuild`](https://github.com/robotology/codyco-superbuild), as the CoDyCo project had finally ended in April 2017 . 
 As the in the past few months we had a few discussion related to "we would really need a general superbuild for the software developed in the iCub Facility", I structured the new superbuild in a way that  it can be easily used for all software developed in the robotology repository. 

The key concept is the "profile" (if you have a better name please propose one), that is a group of software projects that a given user is interested in install (as I would imagine that the average user is not interested in installing **all the software** contained in the robotology organization). 

At the moment the supported profile are: 
* Core : the "core" software (RTF, YARP, ICUB, ..) 
* Dynamics: the software used by DIC/LOC2 researchers, including Simulink-based controllers. 

**TODO: finish : )**

claudiofantacci, 2017-11-25:
Thanks @traversaro for the huge effort on doing this.
I'm more than happy to be of help supporting any or all grasping/vision-related code for the superbuild. I'm more on the spot for the grasping-related code for which I would probably start adding support, but can later add effort on the vision part.

> Port the instructions to compile YARP and ICUB on the pc104/icub-head (http://wiki.icub.org/wiki/Compilation_on_the_pc104 and robotology/icub-main#456) to use the superbuild with appropriate settings, train all the people involved in the maintenance of the robot and eventually migrate the robot setups to use the superbuild. This will simplify refactoring the contents of the Core repository in the future.

YES! 🎉 

> Provide a vcpkg-based installer for all the dependencies (including vision/learning one) of software in the superbuild, making sure that the superbuild can be easily installed even on Windows.

YES 2! 🎉 

I need to deeply understand the superbuild, which will require me some time. But we can start discussing it during the week.

lornat75, 2017-11-27:
Thanks for this, the intended use of the YCM tool was exactly to have a robotology superbuild.

Comment: instead of an explicit profile in CMake why not leave the options generic and populate a set of .cmake with the name of the profile that set options appropriately? e.g.

- IOL.cmake
- ICUB_ROBOT_CAN.cmake

wouldn't this be more maintainable?

traversaro, 2017-11-28:
> Comment: instead of an explicit profile in CMake why not leave the options generic and populate a set of .cmake with the name of the profile that set options appropriately? e.g.
> 
> - IOL.cmake
> - ICUB_ROBOT_CAN.cmake
>
> wouldn't this be more maintainable?

I don't have a strong feeling about this, but the reasons for which I went for explicit option in CMake are: 
* YCM currently does not supports to enable/disable single project to build with a CMake option (i.e. there is no `YCM_EP_BUILD_<package>` option). Adding of emulating such an option could be useful, but was an additional work with an uncertain outcome.  
* Enabling/disabling an option from the CMake GUI on Windows of the `ccmake` command on *nix system is something that all users are familiar with. Passing a initial cmake cache using the `-C` command is error-prone (you need to get the right location), would require the Windows users to use a terminal and is not familiar to most user (I, for one, had to search the right option to do that on Google). 
* Enable/disabling  a CMake option in an existing build is trivial with a CMake option, while I don't know if it is possible with cmake cache files. 
* Composing different profiles (i.e. let's say that you want to compile IOL, Dynamics and R1) is trivial if they are exposed as CMake options (just enable all of them) while it is impossible when using CMake cache files. 

As always, I will be happy to discuss this in person if necessary.

lornat75, 2017-11-28:
your motivations are convincing 👍

diegoferigo, 2017-12-11:
Great work @traversaro, I'm eagerly looking forward to it. Some random comments derived also from my [current workflow](https://github.com/diegoferigo/dockerfiles/blob/2210024c09fe91e8246eb243a634ffd1eccd90e4/Development/Dockerfile):

* It would be nice having the possibility of disabling ACE ([here](https://github.com/traversaro/robotology-superbuild/blob/master/cmake/BuildYARP.cmake#L8))
* It would be nice enabling / disabling GUIs, especially when one of the `*_ROBOT*` option is enabled
* I don't see QT5 dependencies for GUIs
* `icub-models` repository is still missing, and slowly `icub-gazebo` will become legacy

traversaro, 2017-12-14:
> It would be nice having the possibility of disabling ACE (here)

I would be happy to see if we can find an appropriately flexible way of specifyng this kind of options for the subprojects. For the time being I think 95% of the users of the superbuild will be happy to just compile YARP with ACE, so I think it make sense to keep it like that.  

> It would be nice enabling / disabling GUIs, especially when one of the *_ROBOT* option is enabled

This is a good point, I think it could be a nice "dependency" option such as `ROBOTOLOGY_USES_GUIS` or something similar.

> I don't see QT5 dependencies for GUIs

Fixed in https://github.com/traversaro/robotology-superbuild/pull/9 .

> icub-models repository is still missing, and slowly icub-gazebo will become legacy

Created a separate issue in https://github.com/traversaro/robotology-superbuild/issues/14 .