robotology/robotology-superbuild #4: Document how to find homebrew-installed Qt5 in macOS

traversaro, 2017-12-05:
Qt5 is not symlinked in `/usr/local` by default in the homebrew formula, so Qt5_DIR needs to be properly set to make sure that CMake-based projects are able to find Qt5.
See https://github.com/robotology/yarp/blob/b30f2a8ec899d2b1ed9475191cbc3db224ffd9c9/doc/installation/install_mac.dox#L137 . 
cc @claudiofantacci

claudiofantacci, 2017-12-05:
This has always been a majour issue.
I usually symlink QT5 to solve this kind of problem because adding `export Qt5_DIR=/usr/local/opt/qt5/lib/cmake` do not solve CMake issue 100% of the time.

Actually, by my experience, if you don't symlink QT5 you should add the followin:
```
Qt5Widgets_DIR=/usr/local/opt/qt/lib/cmake/Qt5Widgets
Qt5Test_DIR=/usr/local/opt/qt/lib/cmake/Qt5Test
Qt5Core_DIR=/usr/local/opt/qt/lib/cmake/Qt5Core
Qt5_DIR=/usr/local/opt/qt/lib/cmake
Qt5Concurrent_DIR=/usr/local/opt/qt/lib/cmake/Qt5Concurrent
Qt5Gui_DIR=/usr/local/opt/qt/lib/cmake/Qt5Gui
```

traversaro, 2018-03-09:
I was always able to compile everything by just setting `Qt5_DIR`, anyone have a different experience @vtikha @DanielePucci @nunoguedelha @claudia-lat ?

DanielePucci, 2018-03-09:
I usually disable by setting `ICUB_USE_Qt5 = OFF` 🛩

nunoguedelha, 2018-03-09:
Same for me, I usually only set `Qt5_DIR`.

traversaro, 2018-03-09:
> I usually disable by setting `ICUB_USE_Qt5 = OFF` 🛩 

Are you sure? Qt5 is used also in YARP to build `yarpmotorgui`.

DanielePucci, 2018-03-13:
I think so: I usually used the `yarpmotorgui` using the robot computers

claudiofantacci, 2018-03-13:
> I was always able to compile everything by just setting `Qt5_DIR`, anyone have a different experience?

I had some issues with some other packages which I don't recall at this very moment.  
It is something related to some Homebrew formulae, not directly to our software, and how they look for QT5.  
I guess that setting `Qt5_DIR` without symlinking it is sufficient.

lucaTagliapietra, 2018-05-10:
Readme updated. Once the PR is merged, we can close.

traversaro, 2018-05-10:
Fixed by https://github.com/robotology/robotology-superbuild/pull/71 .

closed 2018-05-10 by traversaro

spotanjo3, 2021-05-12:
Hello. I am a macOS user. How do I change mine to this cmake -DCMAKE_BUILD_TYPE=Release -DQt5_DIR=/usr/local/opt/qt/lib/cmake/Qt5 ?

traversaro, 2021-05-12:
Hi @spotanjo3, what do you mean with "How do I change mine" ? Furthermore, I suggest you to open a new issue, commenting on an old issue is problematic as it tends to be lost.