robotology/robotology-superbuild #6: Missing Qt5 related dependencies in README.md

xEnVrE, 2017-12-10:
I am trying `robotology-superbuild` on a fresh Ubuntu 16.04.3 installation and following the instructions found [here](https://github.com/traversaro/robotology-superbuild#linux).

In [`cmake/BuildYARP.cmake`](https://github.com/traversaro/robotology-superbuild/blob/master/cmake/BuildYARP.cmake) `YARP` is configured with the CMake argument 
```
DCREATE_GUIS:BOOL=ON
```
that requires the `Qt5` related dependencies listed [here](http://www.yarp.it/dependencies.html#dependencies_qt5_debian_ubuntu) (at least for Ubuntu/Debian based environments) but not listed in the [System Dependecies](https://github.com/traversaro/robotology-superbuild#linux) section of the README.md.

traversaro, 2017-12-11:
@xEnVrE Thanks for noting this! 
I think we can add all the packages listed in http://www.yarp.it/dependencies.html (including `qml-module-qt-labs-folderlistmodel` and `qml-module-qt-labs-settings` as they are available in all the Debian/Ubuntu version that we support http://wiki.icub.org/wiki/YARP_Supported_Distributions ) to the list in https://github.com/traversaro/robotology-superbuild#linux . If you have time, can your add them with a Pull Requests? Thanks!

xEnVrE, 2017-12-11:
For sure @traversaro!

traversaro, 2017-12-14:
Fixed in https://github.com/traversaro/robotology-superbuild/pull/9 .

closed 2017-12-14 by traversaro