robotology/robotology-superbuild #12: YCM error with CMake 3.2

traversaro, 2017-12-11:
~~~
CMake Error at build/install/share/YCM/cmake-next/Modules/FindPackageHandleStandardArgs.cmake:132 (include):
  include could not find load file:
    /home/travis/build/traversaro/robotology-superbuild/build/install/share/YCM/cmake-next/Modules/FindPackageMessage.cmake
Call Stack (most recent call first):
  build/install/share/YCM/modules/StandardFindModule.cmake:88 (include)
  build/install/share/YCM/find-modules/FindACE.cmake:41 (include)
  cmake/BuildYARP.cmake:8 (find_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:235 (include)
  CMakeLists.txt:105 (find_or_build_package)
-- Performing Test ACE_COMPILES_WITHOUT_INLINE_RELEASE
-- Performing Test ACE_COMPILES_WITHOUT_INLINE_RELEASE - Success
CMake Error at build/install/share/YCM/cmake-next/Modules/FindPackageHandleStandardArgs.cmake:132 (include):
  include could not find load file:
    /home/travis/build/traversaro/robotology-superbuild/build/install/share/YCM/cmake-next/Modules/FindPackageMessage.cmake

Call Stack (most recent call first):
  build/install/share/YCM/3rdparty/FindEigen3.cmake:91 (include)
  cmake/BuildYARP.cmake:10 (find_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:235 (include)
  CMakeLists.txt:105 (find_or_build_package)
~~~
https://travis-ci.org/traversaro/robotology-superbuild/jobs/314834847

@drdanz Any idea? I will probably just use CMake 3.5 in Travis, but I want to let you know about this.

drdanz, 2017-12-12:
Is this master or devel?

traversaro, 2017-12-12:
It is a bit tricky to test because I only have CMake 3.2 on Travis, but the problem occurred on the in the YCM commit `set(YCM_TAG 250b19c08cf99e52e89497418261163197500e44)`.

drdanz, 2017-12-13:
Ok, I found the problem, I couldn't reproduce it because it happened only if `YCM_MAINTAINER_MODE` is not enabled. Please try `set(YCM_TAG f64d712fa06e04ba1611ea3a0b0389071dd849be)`

traversaro, 2018-03-09:
Travis updated their CMake to 3.9, so this is quite difficult to check now, closing.

closed 2018-03-09 by traversaro

drdanz, 2018-03-11:
@traversaro is this the default CMake? Do you have to install/update cmake/cmake3 from apt?

traversaro, 2018-03-11:
It is the default one, see https://docs.travis-ci.com/user/build-environment-updates/2017-12-12/ . I am not sure if it gets installed if you remove the `cmake` package and install the `cmake3` one.

drdanz, 2018-03-12:
I think that in yarp we don't get that, since we are still running the build in `group: deprecated-2017Q3`, I don't remember why we had to add it, I'll make some tests...