robotology/robotology-superbuild #7: Missing folders in "../robotology-superbuild/build/install/"

marco-monforte, 2017-12-11:
I followed the instructions in the README.md and seems everything went fine, I do not have errors during the compilation. But when I tried to configure my environment, I found out that the following folders have not been installed and do not know why:
```
../robotology-superbuild/build/install/lib
../robotology-superbuild/build/install/share/yarp
../robotology-superbuild/build/install/share/iCub
```

traversaro, 2017-12-11:
I suspect that you already have most of the libraries provided in `CORE` profile (`YARP`, `ICUB`, `RTF`) installed on your machine, and so the superbuild is finding the one on your system (see http://robotology.github.io/ycm/gh-pages/master/manual/ycm-superbuild.7.html#ycm-superbuild-manual-for-developers and http://robotology.github.io/ycm/gh-pages/master/module/FindOrBuildPackage.html?highlight=find#find_or_build_package). To force the superbuild to build a given package, you can set the `USE_SYSTEM_<PACKAGE>` CMake option to `FALSE`, either using the CMake gui or passing `-DUSE_SYSTEM_<PACKAGE>:BOOL=FALSE` to the `cmake` command line argument.

traversaro, 2018-03-09:
Closing as I guess the case was the one described in https://github.com/robotology/robotology-superbuild/issues/7#issuecomment-350738061 . @marco-monforte please reopen if necessary, thanks.

closed 2018-03-09 by traversaro