robotology/robotology-superbuild #8: Cross-talking problems with existing superbuilds

traversaro, 2017-12-11:
A lot of users (@GiulioRomualdi, @xEnVrE and also @marco-monforte, see https://github.com/traversaro/robotology-superbuild/issues/7 ) had problem related to the fact that the superbuild finds packages that are already installed in the system, either via another superbuild or by standalone installation. 
Why this is exactly the operating principle of the superbuild, we should cleanly explain this in the readme, to be sure that users are not confused.

traversaro, 2018-03-09:
I think @jeljaik was also confused by this. @aikolina do you have any idea on how to document this in a clear way?

aikolina, 2018-03-09:
@traversaro I can take care of it, but I will probably need someone's help to recall some concepts.

traversaro, 2018-03-09:
Ok, I can provide any detail you will need.

traversaro, 2018-07-11:
Fixed by https://github.com/robotology/robotology-superbuild/pull/99 .

closed 2018-07-11 by traversaro