robotology/robotology-superbuild #13: YARP compilation failure in Travis

traversaro, 2017-12-12:
~~~
In file included from /home/travis/build/traversaro/robotology-superbuild/robotology/YARP/src/libYARP_OS/src/Companion.cpp:7:
In file included from /home/travis/build/traversaro/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/impl/Companion.h:15:
/home/travis/build/traversaro/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/impl/PlatformMap.h:22:48: error: redefinition of 'ACE_Equal_To<std::string>'
class __attribute__ ((visibility ("default"))) ACE_Equal_To<std::string>
                                               ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/ace/Functor_String.h:86:47: note: previous definition is here
class __attribute__ ((visibility("default"))) ACE_Equal_To<std::string>
                                              ^
In file included from /home/travis/build/traversaro/robotology-superbuild/robotology/YARP/src/libYARP_OS/src/Companion.cpp:7:
In file included from /home/travis/build/traversaro/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/impl/Companion.h:15:
/home/travis/build/traversaro/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/impl/PlatformMap.h:31:48: error: redefinition of 'ACE_Hash<std::string>'
class __attribute__ ((visibility ("default"))) ACE_Hash<std::string>
                                               ^~~~~~~~~~~~~~~~~~~~~
/usr/include/ace/Functor_String.h:98:47: note: previous definition is here
class __attribute__ ((visibility("default"))) ACE_Hash<std::string>
                                              ^
In file included from /home/travis/build/traversaro/robotology-superbuild/robotology/YARP/src/libYARP_OS/src/Companion.cpp:7:
In file included from /home/travis/build/traversaro/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/impl/Companion.h:15:
/home/travis/build/traversaro/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/impl/PlatformMap.h:50:48: error: redefinition of 'ACE_Less_Than<std::string>'
class __attribute__ ((visibility ("default"))) ACE_Less_Than<std::string>
                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/ace/Functor_String.h:111:47: note: previous definition is here
class __attribute__ ((visibility("default"))) ACE_Less_Than<std::string>
                                              ^
3 errors generated.
make[5]: *** [src/libYARP_OS/CMakeFiles/YARP_OS.dir/src/Companion.cpp.o] Error 1
make[4]: *** [src/libYARP_OS/CMakeFiles/YARP_OS.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [robotology/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/YARP.dir/all] Error 2
make: *** [all] Error 2
~~~

See https://travis-ci.org/traversaro/robotology-superbuild/jobs/315414081 . 
cc @drdanz Any idea?

traversaro, 2017-12-12:
Related to https://github.com/robotology/yarp/issues/1466 .

traversaro, 2017-12-12:
Apparently you can't compile YARP `master` with YCM `devel` (introduced by https://github.com/traversaro/robotology-superbuild/pull/2). : (

claudiofantacci, 2017-12-12:
> Apparently you can't compile YARP master with YCM devel (introduced by #2). : (

I think it's something very recent, and may not for the next releases.
@drdanz may clarify this.

traversaro, 2017-12-12:
The "cleanest" solution I can think of is to rollback https://github.com/robotology/ycm/commit/dfdb8d875489a0357603ae67b05706f2ffd335c9 .

claudiofantacci, 2017-12-12:
I updated my previous post that was uncomplete. I report it here for completeness:
> I think it's something very recent, and may not be a problem starting for the next release.

As far as
> The "cleanest" solution I can think of is to rollback robotology/ycm@dfdb8d8 .

is concerned, it all dependes on @drdanz.
The idea was to stop using string prvided by ACE and use the std one.

drdanz, 2017-12-13:
> The "cleanest" solution I can think of is to rollback robotology/ycm@dfdb8d8 .

I will not revert that, `ACE_HAS_STRING_HASH` was true in every ACE version that we support (and has been true for a long time). Instead I could add a `set(ACE_HAS_STRING_HASH 1)` with a [`variable_watch`](https://cmake.org/cmake/help/v3.10/command/variable_watch.html) that prints a warning when the variable is used.
@traversaro does this sound ok to you?

traversaro, 2017-12-13:
>  @traversaro does this sound ok to you?

That's great!

drdanz, 2017-12-13:
Done, please try robotology/ycm@85bf58f442ee0e8fce8cafeae314da95ef46172f

traversaro, 2017-12-19:
Fixed by https://github.com/robotology/robotology-superbuild/pull/11 , cc @lucaTagliapietra .

closed 2017-12-19 by traversaro