robotology/robotology-superbuild #14: Add icub-models

traversaro, 2017-12-14:
https://github.com/robotology-playground/icub-models

traversaro, 2018-05-30:
@gabrielenava What is the current status of this? We are still using the models provided by `yarpWholeBodyInterface` ?

gabrielenava, 2018-05-31:
>@gabrielenava What is the current status of this? We are still using the models provided by yarpWholeBodyInterface when whole-body-controllers is installed using the robotology-superbuild ?

this point:

>Test the real robot models present both in yarpWholeBodyInterface and icub-models to make sure that demos work fine with icub-models models. Once this has been assessed, add icub-models to the codyco-superbuild and remove from yarpWholeBodyInterface the models already contained in icub-models .

Has been achieved. The robot was able to perform the Yoga demo in simulation, see .

This other task

> Understand how to handle the icubGazeboSim model, that is a model roughly matching the icub model contained in https://github.com/robotology/icub-gazebo/tree/master/icub .

Is still `TODO`. In particular, `icubGazeboSim` model is still required for the demos that imply the usage of a Gazebo `world`, like the `icub on a chair` or `icub on a seesaw` demo.

The yarpWholeBodyInterface [configuration files](https://github.com/robotology/yarp-wholebodyinterface/blob/master/app/robots/icubGazeboSim/yarpWholeBodyInterface.ini) are instead no more required, because WBToolbox now takes the configuration specified in [Simulink](https://github.com/robotology/whole-body-controllers/blob/master/torque-controllers/momentum-based-yoga/app/robots/iCubGenova04/configRobot.m#L11).

traversaro, 2018-06-12:
> Has been achieved. The robot was able to perform the Yoga demo in simulation, see .

See? Furthermore, in all the recent demonstration have you use the `icub-models` models or the `yarpWholeBodyInterface` one? 

> Is still TODO. In particular, icubGazeboSim model is still required for the demos that imply the usage of a Gazebo world, like the icub on a chair or icub on a seesaw demo.

This is not a big problem. We can start by removing all the non-icubGazeboSim model.urdf from yarpWholeBodyInterface and decide where to store the `icubGazeboSim` `model.urdf` when we remove the yarpWholeBodyInterface from the superbuild.

gabrielenava, 2018-06-12:
> Has been achieved. The robot was able to perform the Yoga demo in simulation, see .

this issue https://github.com/robotology/icub-gazebo-wholebody/issues/19#issuecomment-387456326

I am currently using the `icub-models` models. On the real robot I suspect we are still using the ones coming from the `yarpWholeBodyInterface`.

lrapetti, 2018-06-15:
`icub-model` successfully tested on iCubGenova04 with Yoga++.

@traversaro

traversaro, 2018-06-17:
See https://github.com/robotology/robotology-superbuild/pull/91 .

> As in the documentation and in the `setup.sh` the `<prefix>/share/iCub` directory is listed in the `YARP_DATA_DIRS` before `<prefix>/share/codyco`, the models installed by this repo will be automatically take the precendence w.r.t. to the ones installed by https://github.com/robotology/yarp-wholebodyinterface/ . Because of this, we can first merge this PR and then remove the models that are overlapping between `icub-models` and `yarp-wholebodyinterface` .  

For `icubGazeboSim`, I think it make sense for the `model.urdf` to be installed by the `icub-gazebo`  repo.