robotology/robotology-superbuild #16: Enabling ROBOTOLOGY_USES_OCTAVE throws cmake error

prashanthr05, 2017-12-18:
The following error is thrown while configuring cmake for `yarp-matlab-bindings` in `core` and `iDyntree` in `dynamics`

`Could NOT find Octave (missing: OCTAVE_ROOT_DIR OCTAVE_INCLUDE_DIRS
  OCTAVE_LIBRARIES)`

Available Octave version on the system is `4.0.0`
 
I also noticed that YCM from another super-build directory is being used to configure the necessary setup. However, since the options `USE_SYSTEM_$PROJECT$` are all disabled, the desired cloning, compiling and installing is performed.

traversaro, 2018-03-09:
Sorry @prashanthr05 for reading this only now,  did you solve your problem in the end?

prashanthr05, 2018-03-09:
@traversaro I haven't had an opportunity to use the robotology-superbuild again. So haven't checked on this problem yet.

traversaro, 2018-03-29:
@fiorisi had the same problem, but it turned out that he was missing the `liboctave-dev` Ubuntu package. The installation of that package is now documented in https://github.com/robotology/robotology-superbuild#linux-2 , so I think we can close the issue.

closed 2018-03-29 by traversaro