robotology/robotology-superbuild #18: Debug prints

lucaTagliapietra, 2017-12-19:
~~~
YCM: _dep_component_set = 1, _dep_component = core
RTF: _dep_component_set = 1, _dep_component = core
YARP: _dep_component_set = 1, _dep_component = core
YARP: _dep_component_set = 1, _dep_component = core
ICUB: _dep_component_set = 1, _dep_component = iCub
YARP: _dep_component_set = 1, _dep_component = core
ICUB: _dep_component_set = 1, _dep_component = iCub
iDynTree: _dep_component_set = 1, _dep_component = dynamics
wholeBodyInterface: _dep_component_set = 1, _dep_component = dynamics
YARP: _dep_component_set = 1, _dep_component = core
ICUB: _dep_component_set = 1, _dep_component = iCub
iDynTree: _dep_component_set = 1, _dep_component = dynamics
wholeBodyInterface: _dep_component_set = 1, _dep_component = dynamics
yarpWholeBodyInterface: _dep_component_set = 1, _dep_component = dynamics
~~~

traversaro, 2017-12-19:
Any idea @drdanz ?

drdanz, 2017-12-20:
I forgot some debug message in one of the latest commits.
Fixed in robotology/ycm@e9bfa3968c1141d8e5a416f648b0a862cf857e2e

traversaro, 2017-12-20:
Thanks, great!

closed 2017-12-20 by traversaro