robotology/robotology-superbuild #20: On Windows, changing the YCM_TAG seems to be ineffective

traversaro, 2017-12-20:
If you have a superbuild with a bootstrapped YCM, on Windows changing the YCM_TAG seems to be ineffective, i.e. the YCM is not updated.

drdanz, 2017-12-20:
Perhaps on Windows, the second time CMake is executed, YCM is found even if it is in the superbuild directory. In order to control of this is what is happening, you can check if `YCM_DIR`is set after the second run.

drdanz, 2017-12-20:
Just opened a partially related issue on CMake: https://gitlab.kitware.com/cmake/cmake/issues/17584

traversaro, 2020-04-26:
As far as  I know this  was  not present anymore. If someone is able to reproduce it, feel free to re-open the issue or to open a new issue.

traversaro, 2020-04-26:
> 
> 
> Perhaps on Windows, the second time CMake is executed, YCM is found even if it is in the superbuild directory. In order to control of this is what is happening, you can check if `YCM_DIR`is set after the second run.

This kind  of  behaviour will be anyhow prevented by https://github.com/robotology/robotology-superbuild/pull/389 .

closed 2020-04-26 by traversaro