robotology/robotology-superbuild #21: Specify a desired version in YCMBootstrap

traversaro, 2017-12-20:
In the existing superbuild, we routinely change the `YCM_TAG` option to specify that we require some newer version of ycm to build the superbuild. However, when we check if a YCM is available in the system, we do not  check for its version, so it often happens that the superbuild is not working because the user has an outdated ycm installed in his system. 

This is similar to the problem that applies to generic projects in the superbuild: https://github.com/robotology/codyco-superbuild/issues/161 .

drdanz, 2017-12-20:
If I remember correctly, YCM version for development builds contains the date information, therefore we should be able two use that for checking the version. An alternative is to increase the patch version for the development branch more often...

traversaro, 2017-12-22:
The point I was making it is a bit different. Now that you released YCM v0.6.0, I can switch to it in  https://github.com/robotology/robotology-superbuild/blob/master/CMakeLists.txt#L93 , but I still can't set the minimum desired version in https://github.com/robotology/robotology-superbuild/blob/master/cmake/YCMBootstrap.cmake#L74 . 
I could fix this by modifying the YCMBootstrap file, but this is against all the hash checks.

closed 2019-05-07 by traversaro