robotology/robotology-superbuild #22: CMake fails if ROBOTOLOGY_USES_PYTHON option is enabled

prashanthr05, 2018-01-05:
The following CMakeError is thrown while compiling the superbuild with the `ROBOTOLOGY_USES_PYTHON` cmake option is turned `ON` 

```
Unknown CMake command "swig_add_library". 
```
On investigation with @traversaro it was noted that the `swig_add_library` macro is available in the `UseSWIG.cmake` only, starting from cmake version `3.8.2`, but ycm handles version only until `3.7`

traversaro, 2018-01-05:
YCM is supposed to provided a newer version of the `UseSWIG.cmake` file, but the relative files are not included in `CMAKE_MODULE_PATH`, probably because `3.8` is not present in this loop: https://github.com/robotology/ycm/blob/61d3a992c39bff30f80bc6fb46041588e517ab30/YCMConfig.cmake.in#L54 . @drdanz

traversaro, 2018-01-05:
Added testing on this option in Travis in https://github.com/robotology/robotology-superbuild/pull/23 .

traversaro, 2018-01-05:
Interestingly, Travis is failing for a different problem ( https://travis-ci.org/robotology/robotology-superbuild/jobs/325479020 ):
~~~
[100%] Swig source
/home/travis/build/robotology/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/ConstString.h:192: Warning 389: operator[] ignored (consider using %extend)
/home/travis/build/robotology/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/ConstString.h:193: Warning 389: operator[] ignored (consider using %extend)
/home/travis/build/robotology/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/ConstString.h:273: Warning 321: 'hash' conflicts with a built-in name in python
/home/travis/build/robotology/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/Portable.h:31: Error: Syntax error in input(3).
make[5]: *** [lib/python/yarpPYTHON_wrap.cxx] Error 1
make[4]: *** [bindings/CMakeFiles/_yarp_python.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [robotology/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/YARP.dir/all] Error 2
make: *** [all] Error 2
~~~

traversaro, 2018-01-05:
Interestingly I wrote the Travis to test the Python2 bindings, but YARP's Travis is testing its Python bindings with Python3 ( https://github.com/robotology/yarp/blob/992ec99d95cff898957c0ab6e2c8f7c9eb2fa0a6/.travis.yml#L85 ), but unfortunately RTF does not support Python3 (https://github.com/robotology/robot-testing/issues/37) : ( .

traversaro, 2018-01-05:
The swig error should be solved by https://github.com/robotology/robotology-superbuild/pull/23/commits/b6a24bbe16725d8d99bf98328cef467e09ac071d (unfortunately the logic to check that swig must be at least version 3 is only available in devel).

prashanthr05, 2018-01-05:
```
The swig error should be solved by b6a24bb (unfortunately the logic to check that swig must be at least version 3 is only available in devel).
```

Is this the error related to the Swig Source ?

```
[100%] Swig source
/home/travis/build/robotology/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/ConstString.h:192: Warning 389: operator[] ignored (consider using %extend)
/home/travis/build/robotology/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/ConstString.h:193: Warning 389: operator[] ignored (consider using %extend)
/home/travis/build/robotology/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/ConstString.h:273: Warning 321: 'hash' conflicts with a built-in name in python
/home/travis/build/robotology/robotology-superbuild/robotology/YARP/src/libYARP_OS/include/yarp/os/Portable.h:31: Error: Syntax error in input(3).
make[5]: *** [lib/python/yarpPYTHON_wrap.cxx] Error 1
make[4]: *** [bindings/CMakeFiles/_yarp_python.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [robotology/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[1]: *** [CMakeFiles/YARP.dir/all] Error 2
make: *** [all] Error 2
```
this one ?

traversaro, 2018-01-05:
Yes.

traversaro, 2018-01-05:
The original problem cannot be reproduced in Travis because the default Travis image have CMake 3.5 installed.

traversaro, 2018-05-17:
I had the same issue.

traversaro, 2018-05-17:
I think this is only happening on YARP devel. 
cc @anqingd

closed 2018-05-17 by drdanz