robotology/robotology-superbuild #25: Test different configurations and linux distros

diegoferigo, 2018-01-22:
One of the main targets of a superbuild is to compile flawlessly on many system configurations. Unfortunately the variability of dependencies and their versions is really wide among distributions, and after changes in handled projects it is easy to forget testing all the supported distros (even the oldest).

In my opinion this is one of the cases where docker features are exploited at their best. I'm open for a discussion on how to include its support to our Travis pipeline.

@traversaro

traversaro, 2018-01-22:
I agree, but if in the near future we will power our Jenkins infrastructure I would think that it make sense to keep the testing of all the possible configurations in Jenkins, as the robotology Travis account is quite underpowered at the moment.

traversaro, 2019-03-13:
That is indeed a desiderata, but I guess we can close this issue unless we have some specific target in the near term. What do you think @diegoferigo ?

diegoferigo, 2019-03-13:
Yes let's close this for now, we can reopen it if it will be the case. We are also waiting to have a more clear idea about the future of Travis development, and putting too much effort in it at the moment is not completely worth.

closed 2019-03-13 by diegoferigo