robotology/robotology-superbuild #28: Document the ROBOTOLOGY_USES_PYTHON option

traversaro, 2018-01-30:
Once it is documented, we can remove the `mark_as_advanced`.

traversaro, 2018-01-30:
Original code in the README removed in https://github.com/robotology/robotology-superbuild/pull/31 :

## Python
Support for this dependency is enabled by the `ROBOTOLOGY_USES_PYTHON` CMake option.

### Configuration
**TODO**

### Check the installation
**TODO**

traversaro, 2018-05-17:
@DanielePucci : @anqingd had problems in understanding out to install python bindings of `iDynTree` due to this missing documentation. As Python is quite used in the rl world, it would be nice to fix this documementation.

DanielePucci, 2018-05-17:
Yes, can @anqingd fix this since he went through it?

traversaro, 2020-03-13:
fyi @paolo-viceconte @diegoferigo @Nicogene

traversaro, 2020-03-13:
Related issues and PR: 
* https://github.com/robotology/robotology-superbuild/pull/359
* https://github.com/robotology/robotology-superbuild/issues/364

traversaro, 2020-03-13:
I edited the top issue description to update on the status as of March 2020.

traversaro, 2020-04-26:
>     * Add documentation of the ROBOTOLOGY_USES_PYTHON option in the README, following the structure of other ROBOTOLOGY_USES_* options

This was fixed by https://github.com/robotology/robotology-superbuild/pull/82 .

traversaro, 2020-04-26:
> Make sure that the option is tested on all the platforms for which it is document to work in README

Python is now tested just on Ubuntu. We should enable it also on macOS, and eventually also on Windows (probably using the Python shipped with Visual Studio?) or explicitly document that it is not supported this options on that platforms.

traversaro, 2020-10-11:
> Python is now tested just on Ubuntu. We should enable it also on macOS, and eventually also on Windows (probably using the Python shipped with Visual Studio?) or explicitly document that it is not [...]

For testing Python on macOS and Windows probably the easiest option is to use conda (see https://github.com/robotology/robotology-superbuild/issues/477 and https://github.com/robotology/robotology-superbuild/issues/477).

traversaro, 2021-09-07:
I think documentation of the option is complete, see https://github.com/robotology/robotology-superbuild/blob/master/doc/cmake-options.md#python .

closed 2021-09-07 by traversaro