robotology/robotology-superbuild #33: How to handle the devel branches?

traversaro, 2018-01-30:
In the last two years, more and more repositories in the robotology organization have started to use the YARP model of having two branches, `master` only for bug fixes and `devel` for active developement. It is also quite common that the `devel` branch of a given package is only compatible with the `devel` branch of its robotology dependencies. 

At the moment, the workflow to deal with "devel" branches is quite cumbersome, and is documented here: https://github.com/robotology/robotology-superbuild#handling-the-devel-branch . 

Anyone has any idea on how to streamline this process? 

Some pointers:
* The advr-superbuild use a bash script to simplify the switch from one branch to another: https://github.com/ADVRHumanoids/advr-superbuild/blob/master/scripts/switch-all.sh but I think it has some strict requirements 

cc @diegoferigo @S-Dafarra @gabrielenava

diegoferigo, 2018-01-30:
I had to go through this process just this morning. Yes, it is cumbersome. In my opinion, along the `YCM_EP_DEVEL_MODE_<package_name>` that can be useful for many reason, we could introduce new variables `<package_name>_CHECKOUT_<branch>` which are defaulted to `master` (this should not require turning on the `DEVEL_MODE` for the target repository).

This would allow an easy switch also towards something else than `devel`.

traversaro, 2018-01-30:
> Yes, it is cumbersome. In my opinion, along the YCM_EP_DEVEL_MODE_<package_name> that can be useful for many reason, we could introduce new variables <package_name>_CHECKOUT_BRANCH which are defaulted to master (this should not require turning on the DEVEL_MODE for the target repository).

Fortunately this already exists, it is the `<package>_TAG` option, see http://robotology.github.io/ycm/gh-pages/master/manual/ycm-superbuild.7.html#overriding-parameters for more options. However, setting `<package>_TAG` **does not** switch the existing repo to that branch, but only uses the branch when the repository is created.

S-Dafarra, 2018-01-30:
I would be already happy if that tag has effects when doing ``make update-all``, but I see that we may easily have problems in case of inconsistencies. Maybe a warning may help in this sense.

traversaro, 2018-01-30:
I see, setting the `<package>_TAG` option and having a magical `make switch-all-repos-to-the-branch-I-specified` target would be the ideal situation. This is a bit problematic given how the ExternalProject module works, but it is good to have in mind the ideal interface from the user's point of view.

traversaro, 2018-03-09:
Related: https://github.com/robotology-playground/robotology-superbuild/issues/12 .

closed 2019-10-24 by traversaro