robotology/robotology-superbuild #39: Repositories in devel and superbuild in NON_INTERACTIVE_BUILD type

diegoferigo, 2018-02-06:
I follow the steps:

- Configure the superbuild as `NON_INTERACTIVE_BUILD`
- `make update_all`
- Manually checkout `devel` in some repos
- Reconfigure CMake setting these repos in `DEVEL_MODE`
- Building the superbuild

I obtain the following error:

```
From https://github.com/robotology/codyco-modules
 * branch              master     -> FETCH_HEAD
First, rewinding head to replay your work on top of it...

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'root@244ab5948956.(none)')
No rebase in progress?
CMake Error at /iit/sources/robotology-superbuild/build/robotology/codyco-modules/CMakeFiles/YCMTmp/codyco-modules-gitupdate.cmake:95 (message):
  

  Failed to rebase in:
  '/iit/sources/robotology-superbuild/robotology/codyco-modules/'.

  You will have to resolve the conflicts manually
```

It looks like the superbuild tries to operate on branches anyways.

For the time being, as temporary workaround I set some dummy local git credentials and then I remove them (my system is generic, during its generation does not know any information about the user that is going to use it, for this reason I use a non-interactive build). In this way, the rebase is performed successfully and `devel` get compiled as expected.

traversaro, 2018-02-06:
If the problem only appears in `codyco-modules` I imagine it could be related to some weird escaping of the package name.

diegoferigo, 2018-02-06:
I reported only the error on `codyco-modules`, but it is the same for `yarp-matlab-bindings` as well.

traversaro, 2018-02-06:
`yarp-matlab-bindings` is another project that have a  hyphen (`‐`) in their package name. I suspect that for some reason the `_DEVEL_MODE` variable is ignored for project that have an hyphen in their package name.

traversaro, 2019-03-13:
Update the title to capture the latest updates.

drdanz, 2019-03-14:
Looks like a YCM bug, should we move it to the YCM repo?

traversaro, 2019-03-14:
Ok for me (even if probably it first make sense to try to get the superbuild to work with the latest YCM, so we can verify if this is a problem even in the latest YCM).