robotology/robotology-superbuild #42: Installing system dependencies on Windows using vcpkg

traversaro, 2018-02-13:
At the moment, in the official documentation we recommend to install most of the "System Dependencies" on Windows using the YARP and iCub dependency installers. 

However, expert  users may prefer to use vcpkg. As already discussed in the past, I think in the long term the most maintainable solution is to generated our installer from vcpkg packages (see https://github.com/robotology-playground/vcpkg-binary-installer-generator/issues/2 for a related discussion). However, until we find the time to work on that, I don't think vcpkg is suitable to be used by every user, due to the long time necessary for the compilation of some packages (qt5). 

For the time being, I will just keep on this issue the vcpkg package that provide the necessary dependencies for each superbuild profile. 
**Note: use `--triplet x86-windows` if you want to build packages for 32 bit.**

##  `ROBOTOLOGY_ENABLE_CORE` 
~~~
 ./vcpkg install --triplet x64-windows ace gsl eigen3 opencv freeglut ode sdl2 qt5
~~~

**Note: qt5 now is split in several smaller subpackages, we should identify the one that are actually necessary for YARP/iCub GUIs.**

## `ROBOTOLOGY_ENABLE_IHMC` 
~~~
 ./vcpkg install --triplet x64-windows asio
~~~

## `ROBOTOLOGY_USES_GAZEBO`
Work in progress at https://github.com/traversaro/gazebo-superbuild#windows . 

cc @diegoferigo @fiorisi @claudiofantacci

traversaro, 2018-02-13:
@diegoferigo @fiorisi @claudiofantacci please update and correct if necessary.

traversaro, 2018-02-13:
Based on the packages documented in http://www.yarp.it/dependencies.html#dependencies_qt5_Linux , I *think* it should be sufficient to install the `qt5-declarative` and `qt5-multimedia` vcpkg ports (and transitively their dependencies) to get the YARP GUIs to work, but I never tried.

diegoferigo, 2018-02-13:
Yes thanks, I was going to try exactly the same thing. In my system, I have the following qt-dependent dependencies:
```
libqcustomplot-dev
qtmultimedia5-dev
qtdeclarative5-dev
libqt5opengl5-dev
qttools5-dev
```

closed 2018-03-04 by traversaro

traversaro, 2020-06-04:
For anyone finding this old issue on Google, the updated version of these docs is available at https://github.com/robotology/robotology-superbuild#system-libraries .