robotology/robotology-superbuild #44: Clean WB-Toolbox dependencies

diegoferigo, 2018-02-23:
The 3.0 release of `WB-Toolbox` dropped the `wholeBodyInterface` and `yarpWholeBodyInterface` dependencies. They can be removed from the superbuild.

diegoferigo, 2018-03-30:
@traversaro @gabrielenava I remember we discussed f2f about a dependency which might prevent this drop, was it [mex-wholebodymodel](https://github.com/robotology/mex-wholebodymodel) (which is not included in this superbuild)?

gabrielenava, 2018-04-16:
I think `mex-wholebodymodel` depends upon the `yarpWholeBodyInterface`, on the other hand the repo is currently not used.

traversaro, 2018-04-16:
If you think you can manage the installation of `mex-wholebodymodel` outside the superbuild, I am ok in removing it.

gabrielenava, 2018-04-16:
I'm ok with this

traversaro, 2018-12-19:
Fixed by https://github.com/robotology/robotology-superbuild/pull/125 (I guess).

closed 2018-12-19 by traversaro