robotology/robotology-superbuild #46: Port the README to the Wiki section?

diegoferigo, 2018-03-04:
With the great work of @traversaro, the README has reached a really good status. Too good I'd say to stay in the README file. There are so many subsections and levels that the readability in my opinion is affected.

I'd suggest to keep a minor description in the README and port the instructions in the project's Wiki, where it can be split in a better way.

traversaro, 2018-03-04:
I agree on the splitting if it helps readability (and I suspect it will, but being the author I am not able to judge), but I am not sure if the wiki of the project is the right place. 

The github wiki (https://help.github.com/articles/about-github-wikis/) is effectivly a "sister" git repository, with slightly nicer interface for rendering and editing online the pages. However, being it a separate git repository w.r.t. to the main `robotology-superbuild` git repo, there are these limitations:
* You lose the nice GitHub workflow of "Changes --> PR --> Review --> More Changes --> More Review --> PR Accepted --> Merge". As I think the documentation is integral part of software, I think it should be subject to exactly the same QA process.
* If you do a modification to the repo content that involve a change to the documentation in the wiki, you have to manually track the two modification in synch, and make sure that they are "merged" together. 
* If you fork the repo, you are not forking the wiki, so it is much more compliated to propose a contribution for people without the writing permissions. 

For all this reasons, I think the wiki is more suitable to be a complement to the GitHub issue tracker system rather than a place to store documentation for end-users of the software. 

I would really be much more in favor of splitting the readme in different `.md` files, and generating a fancy website using  mkdocs or similar tools as you did with the wb-toolbox website. 

cc @alecive @francesco-romano  because we are discussing of this on sunday and they love this stuff.

diegoferigo, 2018-03-04:
Having a generator for the documentation would be the best solution, I totally agree with it. And with solutions like `mkdocs`, applying changes would be trivial (even passing through a PR for who does not have write access).

If it means just porting the current status of the README, and if we like the [`mkdocs-material`](https://squidfunk.github.io/mkdocs-material/), as soon as I have half available day I can take care of it. It would be a trivial port after the work done with wb-toolbox. In the meantime, let's wait other feedback.

traversaro, 2018-03-04:
I hope to get a Jenkins jobs soon for the superbuild, so we don't even have to take care of all the travis --> gh-pages stuff.

traversaro, 2018-03-04:
Material as a theme seems great, totally fine for me.

francesco-romano, 2018-03-04:
Some years ago @traversaro convinced me that the Wiki is not the ideal place. He already explained why. 😄

traversaro, 2018-03-05:
A good point about why **not** to use Markdown for documentation: http://ericholscher.com/blog/2016/mar/15/dont-use-markdown-for-technical-docs/ . I still think we should go for markdown given that we use it extensively on GitHub, but it is good information to have.

traversaro, 2019-11-26:
We started offloading some documentation in markdown documents in https://github.com/robotology/robotology-superbuild/tree/master/doc, and I think this is a nice tradeoff between maintainability and readability. Unless somebody want to volunteer to pimp it more or has a more precise definition of done for this issue, I would close it.

diegoferigo, 2019-11-26:
Thank for taking care of that @traversaro.

I know that, given the current status, it would take quite a lot of work, but the best separation in my opinion is dividing Linux / MacOs / Windows in three different documents. After all, if you're on one OS you don't care reading information about other OSs, and having the information all there might just cause confusion because your reading is not linear and you have to jump to sections here and there.

As advantage, the linux section could be separated in two: 1) user setup and 2) icub head.

Instead, general informations about the profiles, typical workflows, etc could stay in the main README. I'm not saying we have to do it now, but it could be marked as future work and help wanted.

traversaro, 2019-11-26:
I would open a separate issue for moving the installation instructions in separate documents, so that the definition of done is clear.

closed 2019-11-26 by traversaro