robotology/robotology-superbuild #55: Add unicycle-footstep-planner

traversaro, 2018-03-17:
cc @S-Dafarra @DanielePucci

S-Dafarra, 2018-05-30:
It is possible I misunderstood the meaning of this issue. I will try to add it soon!

traversaro, 2018-05-30:
Great, modified the title of the issue. 
It would be great to also modify the Dynamics profile documentation to add a "Check your installation" section that would link a simple tutorial on how to run the walking demo in simulation.

traversaro, 2018-06-13:
Now that https://github.com/robotology/capture-point-walking is available, I think it is possible to proceed.

Any volunteer?

S-Dafarra, 2018-06-13:
I can try to deal with the ``Cmake`` stuff, but I would ask @GiulioRomualdi to fill the tutorial :grin:

S-Dafarra, 2018-06-13:
See also https://github.com/robotology/unicycle-footstep-planner/issues/14

traversaro, 2018-06-13:
@S-Dafarra great! Let me know if there are any problem.

GiulioRomualdi, 2018-06-13:
Right now `qpOASES` is compiled by the `robotology-superbuild` only if `ROBOTOLOGY_USES_MATLAB` is set to `ON`
https://github.com/robotology/robotology-superbuild/blob/04e9aee50e4d8bc6601e588be18ec1bc45f2b03b/cmake/BuildqpOASES.cmake#L24-L29
However, it is needed by the `capture-point-walking`  module and this module does not require MATLAB (i.e. it can be installed inside `iCub-head`)

S-Dafarra, 2018-06-13:
Noticed that :wink: Thanks!

traversaro, 2018-06-13:
>  However, it is needed by the capture-point-walking module and this module does not require MATLAB (e.g. it can be installed inside iCub-head)

This will automatically handled if you list `qpOASES` in the `DEPENDS` of some project (such as `capture-point-walking`) for which then `find_or_build_package()` is listed in https://github.com/robotology/robotology-superbuild/blob/04e9aee50e4d8bc6601e588be18ec1bc45f2b03b/CMakeLists.txt#L120 .

S-Dafarra, 2018-06-13:
See https://github.com/robotology/robotology-superbuild/pull/87

traversaro, 2019-03-13:
Fixed by https://github.com/robotology/robotology-superbuild/pull/87 .

closed 2019-03-13 by traversaro