robotology/robotology-superbuild #64: Clearly document the MATLAB PRELOAD problem

traversaro, 2018-04-19:
See https://github.com/robotology/codyco-superbuild/issues/141 .

traversaro, 2018-11-14:
Relevant comment: 
* https://github.com/robotology/wb-toolbox/issues/129#issuecomment-438832755

traversaro, 2019-01-29:
One of the problems is that the actual workaround necessary depends on (at least on Linux, not sure about macOS/:
1) The libstdc++ version (that is related to the Ubuntu version)
2) The matlab version 
Any possible combination may required different workarounds.

traversaro, 2019-05-19:
Note that the libstdc++ version of each given Ubuntu version is known (see https://packages.ubuntu.com/search?keywords=libstdc%2B%2B) while the specific libstdc++ vendored in each Matlab version can be automatically obtained by installing and inspecting the relative Matlab Runtime ( https://it.mathworks.com/products/compiler/matlab-runtime.html ). 
In theory with this two piece of information it should be  possible to document (using a table, for example) for which combination the LD_PRELOAD workaround is necessary, and when it is not.

diegoferigo, 2019-05-21:
I understand that having a table would be ideal, but its maintenance takes time and effort. Since this workaround is no longer required for recent matlab version, what about adding it in a troubleshooting-like section?

traversaro, 2019-05-21:
Yes, documenting this is totally independent from providing the table, and we should not stop in documenting this even if we do not have the table. I just mentioned the table because I had just discovered Matlab Runtime, and to be honest  I was thinking of an automatic script to generate the table (the major problem is that it probably will require ~50 Gb to download all and unpack the version of Matlab Runtime necessary). 

> Since this workaround is no longer required for recent matlab version
Unfortunately, given the nature of this problem as soon as a new Linux/Ubuntu version will come out (20.04, for example), the workaround may be or may not be necessary depending on the used Matlab version, and the `libstdc++` versions used in Ubuntu and in Matlab.

diegoferigo, 2019-05-21:
> I just mentioned the table because I had just discovered Matlab Runtime, and to be honest I was thinking of an automatic script to generate the table

:top:

> as soon as a new Linux/Ubuntu version will come out (20.04, for example), the workaround may be or may not be necessary depending on the used Matlab version

Last time there was an ABI (?) difference between the two versions if I recall. Let's hope it's not going to happen again next time. Btw this can happen with most of the libraries vendored in Matlab right? So far it happened only with that one for our applications.

traversaro, 2020-11-14:
After dealing with this issue for the last 4 year, I just saw on Twitter this tweet: 
https://twitter.com/r2castellotti/status/1327647698577666053

Through that, I discovered that on Debian-based distributions (such as as Ubuntu) there is a [`matlab-support` package](https://packages.ubuntu.com/focal/matlab-support) that you can install via `apt`, and automatically handles the renaming of the libraries vendored by MATLAB to avoid loading conflicts. It probably make sense to document it, and eventually look for a similar solution in macOS . On Windows fortunately nothing should be necessary, thanks to the more sane shared linking model.

S-Dafarra, 2021-01-25:
> After dealing with this issue for the last 4 year, I just saw on Twitter this tweet:
> https://twitter.com/r2castellotti/status/1327647698577666053
> Through that, I discovered that on Debian-based [...]

Indeed, it seems to be solving https://github.com/robotology/wb-toolbox/issues/199#issuecomment-767013858, even if I have graphics issues now 
```
com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :1, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x14b0cf64, isOwner false, <5657b27f, 339bf8fc>[count 0, qsz 0, owner <NULL>]]]
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
	at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
	... 2 more
MATLAB has experienced a low-level graphics error, and may not have drawn correctly.
Read about what you can do to prevent this issue at Resolving Low-Level Graphics Issues then restart MATLAB.
To share details of this issue with MathWorks technical support,
please include this file with your service request.
MATLAB has experienced a low-level graphics error, and may not have drawn correctly.
Read about what you can do to prevent this issue at Resolving Low-Level Graphics Issues then restart MATLAB.
To share details of this issue with MathWorks technical support,
please include this file with your service request.
```

S-Dafarra, 2021-01-25:
> > After dealing with this issue for the last 4 year, I just saw on Twitter this tweet:
> > https://twitter.com/r2castellotti/status/1327647698577666053
> > Through that, I discovered that on Debian-based distributions (such as as Ubuntu) there is a [`matlab-support` package](https://packages.ubuntu.com/focal/matlab-support) that you can install via `apt`, and automatically handles the renaming of the libraries vendored by MATLAB to avoid loading conflicts. It probably make sense to document it, and eventually look for a similar solution in macOS . On Windows fortunately nothing should be necessary, thanks to the more sane shared linking model.
> 
> Indeed, it seems to be solving [robotology/wb-toolbox#199 (comment)](https://github.com/robotology/wb-toolbox/issues/199#issuecomment-767013858), even if I have graphics issues now
> 
> ```
> com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :1, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x14b0cf64, isOwner false, <5657b27f, 339bf8fc>[count 0, qsz 0, owner <NULL>]]]
> 	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
> 	at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
> 	at java.lang.Thread.run(Thread.java:748)
> Caused by: java.lang.NullPointerException
> 	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
> 	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
> 	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
> 	... 2 more
> MATLAB has experienced a low-level graphics error, and may not have drawn correctly.
> Read about what you can do to prevent this issue at Resolving Low-Level Graphics Issues then restart MATLAB.
> To share details of this issue with MathWorks technical support,
> please include this file with your service request.
> MATLAB has experienced a low-level graphics error, and may not have drawn correctly.
> Read about what you can do to prevent this issue at Resolving Low-Level Graphics Issues then restart MATLAB.
> To share details of this issue with MathWorks technical support,
> please include this file with your service request.
> ```

The graphics error got fixed by launching matlab as ``matlab -softwareopengl``. Note that it seems I don't need to edit the ``LD_LIBRARY_PRELOAD`` variable anymore.

traversaro, 2021-06-05:
This doc page on the topic is nice, we could take inspiration: https://github.com/kyamagu/mexopencv/wiki/Troubleshooting-(UNIX)#macos .

traversaro, 2021-12-06:
This is the problem we were discussing today. @Andrea8Testa

traversaro, 2021-12-14:
> The graphics error got fixed by launching matlab as `matlab -softwareopengl`.

Note that this solution may be useful as a workaround to quick launch something, but it is not ideal as it slows down a lot visualization time, especially in complex 3D plots. See https://github.com/robotology/robotology-superbuild/issues/953 for more info.

closed 2023-08-29 by traversaro