robotology/robotology-superbuild #67: Specify YARP options through the superbuild

GiulioRomualdi, 2018-04-20:
I'm using the `robotology-superbuild` to install YARP on Ubuntu 16.04 but I do not find a way to enable internal YARP options through the superbuild (e.g. `CREATE_DEVICE_LIBRARY_MODULES` and `ENABLE_yarpmod_SDLJoypad`)

Is there a way to enable these options by using the superbuild or should I install YARP following the [standard instructions](http://www.yarp.it/install_yarp_linux.html)?

traversaro, 2018-04-21:
The workflow that we typically use for these cases is:
* Build the superbuild as usual 
* Once you have a valid build of the YARP subproject, `cd` in the build directory of the YARP subproject and configure it, using `cmake`, `ccmake` or `cmake-gui`. 
* Enable the options you are interested in, such as `ENABLE_yarpmod_SDLJoypad`. As this point, the option will be saved in the [CMake cache](http://cgold.readthedocs.io/en/latest/tutorials/variables/cache.html), and so they will be respected also if you compile the superbuild as an whole. 
* Note that this strategy will not work for variables that are explicitly set in the [`CMAKE_ARGS`](https://github.com/robotology/robotology-superbuild/blob/e4fa0621db9100e7958be73de28ed950214145e3/cmake/BuildYARP.cmake#L30) argument in `BuildYARP.cmake`. This is because the `CMAKE_ARGS` are passed to the cmake command used for configuring the YARP subproject **every  time** the subproject is configured, overwriting (if present) any value written in cache that is also listed in the `CMAKE_ARGS` (see the [`ExternalProject` module docs](https://cmake.org/cmake/help/latest/module/ExternalProject.html) for more info on this) . Actually this can be easily avoided, I will open an issue on this. 

As you may imagine, this work-flow is not really intuitive/user friendly, so it may be worth to properly document it. Note that this approach is not useful when you want to toggle options that change the dependencies of the project, as discussed for example in https://github.com/robotology/robotology-superbuild/issues/1#issuecomment-351773847 . 

An alternative option is (as you mentioned) building YARP outside the superbuild, and then making sure that the superbuild finds it by properly setting either the `CMAKE_PREFIX_PATH` or the `YARP_DIR` variables. If the superbuild finds a (suitable) external YARP in the call to [`find_or_build_package(YARP)`](https://github.com/robotology/robotology-superbuild/blob/master/CMakeLists.txt#L110), it will just use the external one and will not download/build its own YARP. The usage of an external YARP can be enforced by setting the superbuild CMake cache variable `USE_SYSTEM_YARP` to `ON`, see https://robotology.github.io/ycm/gh-pages/latest/module/FindOrBuildPackage.html .

I do not think that this steps are currently well exposed in the existing `robotology-superbuild` documentation. Any PR proposing how to document this, for example integrating part of this answer in the existing docs, is **highly welcome**!

traversaro, 2019-03-13:
Renamed the issue to reflect what needs to be done to solve it.

traversaro, 2019-12-09:
Related YCM issue: https://github.com/robotology/ycm/issues/4 .

traversaro, 2020-04-05:
Hi @GiulioRomualdi , 
   I proposed a possible solution to this in https://github.com/robotology/ycm/pull/329 . If that proposal is accepted, then you  can pass to the cmake configuration of the robotology-superbuild an option that species the CMake arguments that are passed to all projects, for example in your case: 
~~~
cmake -DYCM_EP_ADDITIONAL_CMAKE_ARGS:STRING="-DENABLE_yarpmod_SDLJoypad:BOOL=ON"
~~~
This  parameters are passed to all CMake projects of the superbuild (as it is useful for some options, for example `-DBUILD_TESTING:BOOL=ON`), but for project-specific options that should be ok  as long as they are not options  naming conflicts. Do you think that this  solution would be a good solution for your issue?

closed 2020-10-11 by traversaro