robotology/robotology-superbuild #68: Cleanup usage of CMAKE_ARGS and CMAKE_CACHE_ARGS in Build**.cmake files

traversaro, 2018-04-21:
It is necessary to cleanup the usage of the `CMAKE_ARGS` and `CMAKE_CACHE_ARGS` in the calls to `ycm_ep_helper` in `Build***.cmake` files. 

Back in ~2014, I remember that we always had  `CMAKE_CACHE_ARGS`, but then we started adding subproject CMake options that depended on superbuild options, and so we had problem when switching this option on and off that the relative change was not correctly propagated to the subproject (as the initial cache was already populated, see https://github.com/robotology/ycm/issues/22 and https://github.com/robotology/codyco-superbuild/issues/5). From that point, we basically randomly used  `CMAKE_ARGS` and `CMAKE_CACHE_ARGS` until we reached a configuration that was working for most users. 

Given the discussion in https://github.com/robotology/robotology-superbuild/issues/67#issuecomment-383281344 , I think it make sense to review all existing `Build***.cmake` files to make sure that:
* `CMAKE_ARGS` is only used for subproject options that depend on superbuild options 
* `CMAKE_CACHE_ARGS` is used for all other variables 

A more corner-case but still typical use of  `CMAKE_ARGS` is to make sure that a new modification to the `Build***.cmake`  is also propagated correctly to all existing superbuild installations.

traversaro, 2018-04-21:
cc @diegoferigo

drdanz, 2018-05-02:
I'm don't remember if the `ExternalProject` version in YCM supports it, but there is also the `CMAKE_CACHE_DEFAULT_ARGS` option, that should be used for cache arguments that could be changed by the user...

traversaro, 2018-05-20:
Note: in https://github.com/robotology/robotology-superbuild/pull/76 I added the option `-DICUB_INSTALL_WITH_RPATH:BOOL=ON` to `CMAKE_ARGS` because I wanted it to be propagated also to existing builds. However, once some time have passed, I think we can migrate it to `CMAKE_CACHE_ARGS` .

traversaro, 2018-05-20:
@drdanz You are right, given https://gitlab.kitware.com/cmake/cmake/commit/98cdb658e481a23c48ad894bec102ec5f060d72e?view=parallel `CMAKE_ARGS` and `CMAKE_CACHE_ARGS` are actually quite similar, while `CMAKE_CACHE_DEFAULT_ARGS` is the option I was actually thinking of. I updated the issue comments to reflect that.

traversaro, 2019-02-25:
cc @claudiofantacci

traversaro, 2019-07-01:
Relevant comment: https://github.com/robotology/robotology-superbuild/issues/176#issuecomment-507272567 .

traversaro, 2023-10-30:
I just got hit by this. I think we can just switch the few scripts that are using `CMAKE_CACHE_ARGS` to use `CACHE_ARGS`, and do what is described in https://github.com/robotology/robotology-superbuild/issues/68#issue-316481244 if it turns out to be really needed.

closed 2023-10-30 by traversaro