robotology/robotology-superbuild #72: iCub_SIM not found

lrapetti, 2018-05-11:
I have tried to install the robotology-superbuild in Ubuntu 16.04 with the standard configuration, and following the [Installation instructions](https://github.com/robotology/robotology-superbuild/blob/master/README.md). YARP was working correctly, while iCub_SIM was not found.

It turned out that the package libsdl1.2-dev is missing in the [dependencies](https://github.com/robotology/robotology-superbuild/edit/master/README.md#L86). I have installed the missing package and rebuilt ICUB, but again YARP works fine, while when I run ICub_SIM I get the error:
```
iCub_SIM: error while loading shared libraries: libYARP_dev.so.1: cannot open shared object file: No such file or directory
```

anqingd, 2018-05-19:
I got the same error for ```iCub_SIM```:
```iCub_SIM: error while loading shared libraries: libYARP_dev.so.1: cannot open shared object file: No such file or directory```
similar error also for ```yarpmotorgui```
```yarpmotorgui: error while loading shared libraries: libYARP_init.so.1: cannot open shared object file: No such file or directory```

traversaro, 2018-05-20:
@lrapetti sorry, I did not notice that also after building it you had problems in running `iCub_SIM` . I think the problem is that `icub-main` does not enable rpath by default on Linux and macOS . As this is a general possibilities in projects build by the superbuild, I think we should revive https://github.com/robotology/robotology-superbuild/pull/34 .

@anqingd your problem seems to be different, because it also affects `yarpmotorgui` that it is build by YARP, I guess there is something strange in your enviroment related to anaconda as already discussed in https://github.com/robotology/idyntree/issues/441 .

traversaro, 2018-05-20:
https://github.com/robotology/robotology-superbuild/pull/76 enables rpath in ICUB, and solves the problem in this case. However, whenever we will build in the superbuild a project without an easy option to enable rpath, I think it would make sense to revive https://github.com/robotology/robotology-superbuild/pull/34 .

anqingd, 2018-05-20:
> @anqingd your problem seems to be different, because it also affects yarpmotorgui that it is build by YARP, I guess there is something strange in your enviroment related to anaconda as already discussed in robotology/idyntree#441 .

Actually I tried `iCub_SIM`/`yarpmotorgui` after deleting anaconda. I just updated my repo and re-built `robotology-superbuild` the issue remains.

lrapetti, 2018-05-20:
My problem was solved by adding the _LD_LIBRARY_PATH_

closed 2018-05-20 by traversaro