robotology/robotology-superbuild #81: WBToolbox not building in robotology-superbuild

nunoguedelha, 2018-06-03:
### Description
When compiling the superbuild, I get the error:
```
...
    cd /Users/nunoguedelha/dev/robotology-superbuild/robotology/WBToolbox
    export LANG=en_US.US-ASCII
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x c++ -arch x86_64 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -Wno-trigraphs -fpascal-strings -O3 -Wno-missing-field-initializers -Wno-missing-prototypes -Wno-return-type -Wno-non-virtual-dtor -Wno-overloaded-virtual -Wno-exit-time-destructors -Wno-missing-braces -Wparentheses -Wswitch -Wno-unused-function -Wno-unused-label -Wno-unused-parameter -Wno-unused-variable -Wunused-value -Wno-empty-body -Wno-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wno-constant-conversion -Wno-int-conversion -Wno-bool-conversion -Wno-enum-conversion -Wno-float-conversion -Wno-non-literal-null-conversion -Wno-objc-literal-conversion -Wno-shorten-64-to-32 -Wno-newline-eof -Wno-c++11-extensions -DCMAKE_INTDIR=\"Release\" -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -fasm-blocks -fstrict-aliasing -Wdeprecated-declarations -Winvalid-offsetof -mmacosx-version-min=10.12 -Wno-sign-conversion -Wno-infinite-recursion -Wno-move -Wno-comma -Wno-block-capture-autoreleasing -Wno-strict-prototypes -Wno-range-loop-analysis -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/WBToolbox/lib/Release/include -I/Users/nunoguedelha/dev/robotology-superbuild/robotology/WBToolbox/toolbox/autogenerated/include -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/WBToolbox/toolbox/autogenerated/WB-Toolbox.build/Release/ClockServer.build/DerivedSources/x86_64 -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/WBToolbox/toolbox/autogenerated/WB-Toolbox.build/Release/ClockServer.build/DerivedSources -Wmost -Wno-four-char-constants -Wno-unknown-pragmas -F/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/WBToolbox/lib/Release -DNDEBUG -fPIC -std=gnu++11 -MMD -MT dependencies -MF /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/WBToolbox/toolbox/autogenerated/WB-Toolbox.build/Release/ClockServer.build/Objects-normal/x86_64/ClockServer.d --serialize-diagnostics /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/WBToolbox/toolbox/autogenerated/WB-Toolbox.build/Release/ClockServer.build/Objects-normal/x86_64/ClockServer.dia -c /Users/nunoguedelha/dev/robotology-superbuild/robotology/WBToolbox/toolbox/autogenerated/src/thrift/ClockServer.cpp -o /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/WBToolbox/toolbox/autogenerated/WB-Toolbox.build/Release/ClockServer.build/Objects-normal/x86_64/ClockServer.o
In file included from /Users/nunoguedelha/dev/robotology-superbuild/robotology/WBToolbox/toolbox/autogenerated/src/thrift/ClockServer.cpp:4:
/Users/nunoguedelha/dev/robotology-superbuild/robotology/WBToolbox/toolbox/autogenerated/include/thrift/ClockServer.h:7:10: fatal error: 'yarp/os/Wire.h' file not found
#include <yarp/os/Wire.h>
         ^~~~~~~~~~~~~~~~
1 error generated.

** BUILD FAILED **

```
So, the header file `yarp/os/Wire.h` can't be found but it is there, in `.../robotology-superbuild/build/install` folder.

### Analysis
The include path `-isystem <robotology-superbuild-root>/build/install/include` is missing in the compile command. This sould be due to some wrong configuration in the cmake files of WBToolbox imported repo.

traversaro, 2018-06-03:
Are you using the `devel` branches or `master` of everything? 

cc @diegoferigo

nunoguedelha, 2018-06-03:
I'm using the devel branches for the below components:
- YCM_EP_DEVEL_MODE_GazeboYARPPl
- YCM_EP_DEVEL_MODE_ICUB
- YCM_EP_DEVEL_MODE_WBToolbox
- YCM_EP_DEVEL_MODE_YARP
- YCM_EP_DEVEL_MODE_codyco-modules
- YCM_EP_DEVEL_MODE_iDynTree
- YCM_EP_DEVEL_MODE_yarp-matlab-bindings

That shouldn't be related to the fact that the include path is missing from the build command should it?

nunoguedelha, 2018-06-04:
@diegoferigo I might be wrong, but I see something suspicious in this [line](https://github.com/robotology/wb-toolbox/blob/7e1766733287f76ce967dbf0e202cd0ab51e846d/toolbox/CMakeLists.txt#L14) of the toolbox `CMakelists.txt`. It's including the file `YarpInstallationHelpers.cmake`, which appears to be out of scope since the only `.cmake` files present in the repo are : 
- AddGeneratedCodeLib.cmake
- ExtraPackageConfigVars.cmake.in
- Utilities.cmake
I guess that `ExtraPackageConfigVars.cmake.in` should expand to `@CMAKE_INSTALL_PREFIX@/share/WB-Toolbox/cmake/*` but the folder doesn't by the time we run the WBToolbox build.
Or maybe just a `find_package(YARP)` is missing somewhere...

diegoferigo, 2018-06-04:
@nunoguedelha 

> I guess that ExtraPackageConfigVars.cmake.in should expand to @CMAKE_INSTALL_PREFIX@/share/WB-Toolbox/cmake/* but the folder doesn't by the time we run the WBToolbox build.
Or maybe just a find_package(YARP) is missing somewhere...

Just tried on a clean superbuild folder and it works fine. For what concern the variable expansion I correctly get something similar

```cmake
#####################################################################
#### Expanded from INCLUDE_FILE by install_basic_package_files() ####

list(APPEND CMAKE_MODULE_PATH /tmp/robotology-superbuild/build/install/share/WB-Toolbox/cmake)

#####################################################################
```

> I might be wrong, but I see something suspicious in this line of the toolbox CMakelists.txt. It's including the file YarpInstallationHelpers.cmake, which appears to be out of scope

That file is part of YCM, which is a dependency

I couldn't test it properly due to https://github.com/robotology/yarp/issues/1722 which prevents me to build the latest version of Yarp. Are you aligned on the last devel commit of Yarp?

nunoguedelha, 2018-06-04:
> I couldn't test it properly due to robotology/yarp#1722 which prevents me to build the latest version of Yarp. Are you aligned on the last devel commit of Yarp?

@diegoferigo I'm building with that same commit but I don't get the same error as you.

> Digging a bit more, the error looks a missing include directory of Yarp for the ClockServer target. But here you can see that it is already included:

Yes, as I described in the issue, the missing header (at least in my environment), is the file `yarp/os/Wire.h`, and it's because **the include path `-isystem <robotology-superbuild-root>/build/install/include` is missing in the compile command**.

nunoguedelha, 2018-06-04:
For the record...
As @diegoferigo had suggested, I logged the value of `YARP_INCLUDE_DIRS`, and as expected it was empty and we get this deprecation warning:
```
CMake Deprecation Warning at /Users/nunoguedelha/dev/robotology-superbuild/build/install/lib/cmake/YARP/YARPConfig.cmake:212 (message):
  The YARP_INCLUDE_DIRS variable is deprecated
Call Stack (most recent call first):
  toolbox/autogenerated/CMakeLists.txt:9999 (deprecated_variable_warning)
  toolbox/autogenerated/CMakeLists.txt:35 (target_include_directories)
```
Which is expected since `YARP_INCLUDE_DIRS` is no more defined and exported by **Yarp** (refer to `<robotology-superbuild-root>/build/install/lib/cmake/YARP/YARPConfig.cmake`) and has been deprecated (https://github.com/robotology/QA/issues/233). 

So the solution turned to be : 
- either link the libraries of `YARP::YARP_OS` module as suggested by @diegoferigo , by changing the code in the `CMakelists.txt` for the target `ClockServer` as follows, from : 
```
# YARP
find_package(YARP REQUIRED)

# Setup the include directories
target_include_directories(ClockServer PUBLIC
                           ${YARP_INCLUDE_DIRS}
                           $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>)
```
... to...
```
# YARP
find_package(YARP REQUIRED)
target_link_libraries(ClockServer YARP::YARP_OS)
 
# Setup the include directories
target_include_directories(ClockServer PUBLIC
                           $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>)
```

- either redefining the `YARP_INCLUDE_DIRS` through `get_target_property` which retrieves the interface headers path from the given module (in this case `YARP::YARP_OS`), as follows : 
```
get_target_property(YARP_INCLUDE_DIRS YARP::YARP_OS INTERFACE_INCLUDE_DIRECTORIES)
```

I'll submit the PRs to the `WBToolbox` and `yarp-matlab-bindings`. **Thanks a lot guys for the help !!**

diegoferigo, 2018-06-04:
Glad we solved. Do not bother to open a PR to wb-toolbox, I already have those changes staged. Thanks!

nunoguedelha, 2018-06-04:
Fixed by:
- https://github.com/robotology-playground/yarp-matlab-bindings/pull/25
- https://github.com/robotology/wb-toolbox/pull/130

PRs under review.

closed 2018-06-10 by nunoguedelha