robotology/robotology-superbuild #83: Using Ninja generator fails with bad $-escape

diegoferigo, 2018-06-11:
The full error is the following:

```
$ cmake --build .
ninja: error: build.ninja:477: bad $-escape (literal $ must be written as $$)
```

The error is generated by variables such as `$(COLOR)` which should be `$$(COLOR)` or `${COLOR}`. This issue is just a reminder if any of you encounter this issue.

Follow the upstream issue at https://gitlab.kitware.com/cmake/cmake/issues/18062.

traversaro, 2018-06-11:
CMake and ninja version?

diegoferigo, 2018-06-11:
Ubuntu 18:04 default:

* CMake 3.10.2
* Ninja 1.8.2

traversaro, 2019-05-19:
If YCM is bootstrapped, apparently the ninja build also fails with the following error: 
~~~
FAILED: robotology/YCM/CMakeFiles/YCMStamp/YCM-configure 
cd /home/straversaro/src/robotology-superbuild/build/robotology/YCM && /usr/bin/cmake --no-warn-unused-cli "-DCMAKE_PREFIX_PATH:PATH=/home/straversaro/src/robotology-superbuild/build/install;/home/straversaro/src/idjl-software/build/install" -C/home/straversaro/src/robotology-superbuild/build/robotology/YCM/CMakeFiles/YCMTmp/YCM-cache-Release.cmake -GNinja /home/straversaro/src/robotology-superbuild/robotology/YCM && /usr/bin/cmake -E touch /home/straversaro/src/robotology-superbuild/build/robotology/YCM/CMakeFiles/YCMStamp/YCM-configure
Not searching for unused variables given on the command line.
loading initial cache file /home/straversaro/src/robotology-superbuild/build/robotology/YCM/CMakeFiles/YCMTmp/YCM-cache-Release.cmake
CMake Error: Error: generator : Ninja
Does not match the generator used previously: Unix Makefiles
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a different binary directory.
~~~

fbiazin, 2020-01-10:
FIY: I was able to compile with ninja when, after the error, I went to the directory where Gentoo put the related files and replaced all instances of $ with $$ on the file build.ninja 

I don't know the proccess that generate this file, I just saw that it was used to write some parameters and replaced it. I also don't know anything about YCM.

The ebuild (file that controls building on Gentoo) was from 2018 and the directory of the file was /var/tmp/portage/games-engines/devilutionx-9999/work/devilutionx-9999_build then I used the command ebuilt with compile, install and qmerge to continue from where the error happened.

traversaro, 2020-04-26:
> If YCM is bootstrapped, apparently the ninja build also fails with the following error:
>
> ```
> FAILED: robotology/YCM/CMakeFiles/YCMStamp/YCM-configure
> cd [...]

This problem was actually similar to https://github.com/robotology/robotology-superbuild/issues/265, and that was fixed by https://github.com/robotology/robotology-superbuild/pull/276 .

However, I am not sure this is related at all with the original issue reported by @diegoferigo . 

However, @diegoferigo are you still experiencing this error? We currently have CI jobs that use Ninja with no problems, so unless there is a way to reproduce this error, I would not know how to tackle it.

diegoferigo, 2020-04-26:
Nope, I don't see this error in a while now (the first post is 2y+ ago). I think we can close this issue and reopen it if the problem shows up again in the future.

Brouilles, 2020-06-27:
Hi, 
I have this issue with CMake and Ninja on Windows 10 and VS Community 2019 (last version). 
Any idea?

Thanks!

giotherobot, 2024-01-25:
This problem resurfaced while trying a fresh superbuild on my laptop and building with VSCode (which uses ninja as defauls instead of unix makefiles).

This seems related to enabling any of the `YCM_EP_DEVEL_MODE_` flags, when none of these are are active the build finshes successfully.

traversaro, 2024-08-13:
Related issue: https://github.com/ninja-build/ninja/issues/1923 . 

I replicated the issue locally by setting `YCM_EP_DEVEL_MODE_ICUB`:
~~~
CMake Error:
  Running

   '/home/traversaro/robotology-superbuild/.pixi/envs/default/bin/ninja' '-C' '/home/traversaro/robotology-superbuild/.build' '-t' 'recompact'

  failed with:

   ninja: error: build.ninja:1114: bad $-escape (literal $ must be written as $$)
~~~

This is the diff in the build.ninja when the `YCM_EP_DEVEL_MODE_ICUB` option is enabled:

~~~diff
(robotology-superbuild) traversaro@IITBMP014LW012:~/robotology-superbuild/.build$ diff build.ninja buildnotworking.ninja
221c221
< build update-all: phony ICUB-update ICUBcontrib-update YARP-update YCM-update ergocub-software-update gz-sim-yarp-plugins-update icub-models-update robots-configuration-update yarp-devices-ros-update
---
> build update-all: phony ICUBcontrib-update YARP-update YCM-update ergocub-software-update gz-sim-yarp-plugins-update icub-models-update robots-configuration-update yarp-devices-ros-update
275c275
< build ICUB: phony CMakeFiles/ICUB CMakeFiles/ICUB-complete src/ICUB/CMakeFiles/YCMStamp/ICUB-done src/ICUB/CMakeFiles/YCMStamp/ICUB-build src/ICUB/CMakeFiles/YCMStamp/ICUB-configure src/ICUB/CMakeFiles/YCMStamp/ICUB-download src/ICUB/CMakeFiles/YCMStamp/ICUB-install src/ICUB/CMakeFiles/YCMStamp/ICUB-mkdir src/ICUB/CMakeFiles/YCMStamp/ICUB-patch_disconnected src/ICUB/CMakeFiles/YCMStamp/ICUB-update_disconnected CMakeFiles/YARP-complete src/YARP/CMakeFiles/YCMStamp/YARP-done CMakeFiles/YCM-complete src/YCM/CMakeFiles/YCMStamp/YCM-done src/YARP/CMakeFiles/YCMStamp/YARP-build src/YARP/CMakeFiles/YCMStamp/YARP-configure src/YARP/CMakeFiles/YCMStamp/YARP-download src/YARP/CMakeFiles/YCMStamp/YARP-install src/YARP/CMakeFiles/YCMStamp/YARP-mkdir src/YARP/CMakeFiles/YCMStamp/YARP-patch_disconnected src/YARP/CMakeFiles/YCMStamp/YARP-update_disconnected src/YCM/CMakeFiles/YCMStamp/YCM-build src/YCM/CMakeFiles/YCMStamp/YCM-configure src/YCM/CMakeFiles/YCMStamp/YCM-download src/YCM/CMakeFiles/YCMStamp/YCM-install src/YCM/CMakeFiles/YCMStamp/YCM-mkdir src/YCM/CMakeFiles/YCMStamp/YCM-patch_disconnected src/YCM/CMakeFiles/YCMStamp/YCM-update_disconnected ICUB-download YARP YCM
---
> build ICUB: phony CMakeFiles/ICUB CMakeFiles/ICUB-complete src/ICUB/CMakeFiles/YCMStamp/ICUB-done src/ICUB/CMakeFiles/YCMStamp/ICUB-build src/ICUB/CMakeFiles/YCMStamp/ICUB-configure src/ICUB/CMakeFiles/YCMStamp/ICUB-download src/ICUB/CMakeFiles/YCMStamp/ICUB-install src/ICUB/CMakeFiles/YCMStamp/ICUB-mkdir src/ICUB/CMakeFiles/YCMStamp/ICUB-patch_disconnected src/ICUB/CMakeFiles/YCMStamp/ICUB-update_disconnected CMakeFiles/YARP-complete src/YARP/CMakeFiles/YCMStamp/YARP-done CMakeFiles/YCM-complete src/YCM/CMakeFiles/YCMStamp/YCM-done src/YARP/CMakeFiles/YCMStamp/YARP-build src/YARP/CMakeFiles/YCMStamp/YARP-configure src/YARP/CMakeFiles/YCMStamp/YARP-download src/YARP/CMakeFiles/YCMStamp/YARP-install src/YARP/CMakeFiles/YCMStamp/YARP-mkdir src/YARP/CMakeFiles/YCMStamp/YARP-patch_disconnected src/YARP/CMakeFiles/YCMStamp/YARP-update_disconnected src/YCM/CMakeFiles/YCMStamp/YCM-build src/YCM/CMakeFiles/YCMStamp/YCM-configure src/YCM/CMakeFiles/YCMStamp/YCM-download src/YCM/CMakeFiles/YCMStamp/YCM-install src/YCM/CMakeFiles/YCMStamp/YCM-mkdir src/YCM/CMakeFiles/YCMStamp/YCM-patch_disconnected src/YCM/CMakeFiles/YCMStamp/YCM-update_disconnected ICUB-configure ICUB-download YARP YCM
279c279
< # Utility command for ICUB-update
---
> # Utility command for ICUB-configure
281c281
< build ICUB-update: phony CMakeFiles/ICUB-update src/ICUB/CMakeFiles/YCMStamp/ICUB-update src/ICUB/CMakeFiles/YCMStamp/ICUB-download src/ICUB/CMakeFiles/YCMStamp/ICUB-mkdir ICUB-download
---
> build ICUB-configure: phony CMakeFiles/ICUB-configure src/ICUB/CMakeFiles/YCMStamp/ICUB-configure src/ICUB/CMakeFiles/YCMStamp/ICUB-patch_disconnected CMakeFiles/YARP-complete src/YARP/CMakeFiles/YCMStamp/YARP-done CMakeFiles/YCM-complete src/YCM/CMakeFiles/YCMStamp/YCM-done src/ICUB/CMakeFiles/YCMStamp/ICUB-update_disconnected src/YARP/CMakeFiles/YCMStamp/YARP-build src/YARP/CMakeFiles/YCMStamp/YARP-configure src/YARP/CMakeFiles/YCMStamp/YARP-download src/YARP/CMakeFiles/YCMStamp/YARP-install src/YARP/CMakeFiles/YCMStamp/YARP-mkdir src/YARP/CMakeFiles/YCMStamp/YARP-patch_disconnected src/YARP/CMakeFiles/YCMStamp/YARP-update_disconnected src/YCM/CMakeFiles/YCMStamp/YCM-build src/YCM/CMakeFiles/YCMStamp/YCM-configure src/YCM/CMakeFiles/YCMStamp/YCM-download src/YCM/CMakeFiles/YCMStamp/YCM-install src/YCM/CMakeFiles/YCMStamp/YCM-mkdir src/YCM/CMakeFiles/YCMStamp/YCM-patch_disconnected src/YCM/CMakeFiles/YCMStamp/YCM-update_disconnected src/ICUB/CMakeFiles/YCMStamp/ICUB-download src/ICUB/CMakeFiles/YCMStamp/ICUB-mkdir ICUB-download YARP YCM
290a291,356
> # Utility command for fetch-all
>
> build fetch-all: phony ICUB-fetch
>
>
> #############################################
> # Utility command for ICUB-fetch
>
> build ICUB-fetch: phony CMakeFiles/ICUB-fetch src/ICUB/CMakeFiles/YCMStamp/ICUB-fetch src/ICUB/CMakeFiles/YCMStamp/ICUB-download src/ICUB/CMakeFiles/YCMStamp/ICUB-mkdir ICUB-download YARP YCM
>
>
> #############################################
> # Utility command for status-all
>
> build status-all: phony ICUB-status
>
>
> #############################################
> # Utility command for ICUB-status
>
[... cut ...]
> #############################################
> # Custom command for src/ICUB/CMakeFiles/YCMStamp/ICUB-dependees-update
>
> build src/ICUB/CMakeFiles/YCMStamp/ICUB-dependees-update | ${cmake_ninja_workdir}src/ICUB/CMakeFiles/YCMStamp/ICUB-dependees-update: CUSTOM_COMMAND || YARP YARP-download YARP-update YCM YCM-download YCM-update
>   COMMAND = cd /home/traversaro/robotology-superbuild/.build && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E echo_append
>   DESC = Dependencies for 'ICUB' updated.
~~~

traversaro, 2024-08-13:
The problematic parts are the `--switch=$(COLOR)` arguments passed to CMake by ninja:

~~~
>   COMMAND = cd /home/traversaro/robotology-superbuild/src/ICUB && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E cmake_echo_color --switch=$(COLOR) --cyan "Working directory: /home/traversaro/robotology-superbuild/src/ICUB" && /usr/bin/git status
>   COMMAND = cd /home/traversaro/robotology-superbuild/src/ICUB && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E echo "" && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E cmake_echo_color --switch=$(COLOR) --cyan "ICUB REPOSITORY directory: " && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E echo "    /home/traversaro/robotology-superbuild/src/ICUB" && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E echo && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E cmake_echo_color --switch=$(COLOR) --cyan "ICUB SOURCE directory: " && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E echo "    " && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E echo "" && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E cmake_echo_color --switch=$(COLOR) --cyan "ICUB BINARY directory: " && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E echo "    /home/traversaro/robotology-superbuild/.build/src/ICUB" && /home/traversaro/robotology-superbuild/.pixi/envs/default/bin/cmake -E echo ""
~~~

traversaro, 2024-08-13:
A fix for this is provided in https://github.com/robotology/ycm-cmake-modules/pull/456 .

closed 2024-08-13 by traversaro