robotology/robotology-superbuild #88: Test usage of robotology-superbuild to compile all necessary software on the real robot

traversaro, 2018-06-15:
robotology-superbuild has been used for a long time on the real robot for  compiling dynamics-related software, especially on the iCub robots `iCubGenova02` and `iCubGenova04` . However, is has always been used just to compile dynamics-specific software, such as `iDynTree` and the `WB-Toolbox` . In this issue we will track the effort necessary to use it to manage **all the software** used in the robot, including `YARP` and `ICUB` . 

**Note: this is an experimental protocol that will be tested for now only on iCubGenova04. If you are a user of any other iCub, please stick to the canonical installation layout described in http://wiki.icub.org/wiki/The_Linux_on_the_pc104 and http://wiki.icub.org/wiki/ICub_server_laptop_installation_instructions.**

traversaro, 2018-06-15:
I have created a copy of all the documentation pages on the wiki to migrate them to the use of robotology-superbuild : 

| Normal page            |  Superbuild version           |
|:-------------------------:|:---------------------------------:|
| http://wiki.icub.org/wiki/ICub_Software_Installation |  http://wiki.icub.org/wiki/ICub_Software_Installation_(superbuild) | 
| http://wiki.icub.org/wiki/Windows:_installation_from_sources  | http://wiki.icub.org/wiki/Windows:_installation_from_sources_using_the_robotology-superbuild | 
|  http://wiki.icub.org/wiki/Linux:Installation_from_sources  |  http://wiki.icub.org/wiki/Linux:Installation_from_sources_using_the_robotology-superbuild  | 
| http://wiki.icub.org/wiki/MacOSX:_installation  |  http://wiki.icub.org/wiki/MacOS:Installation_from_sources_using_the_robotology-superbuild |

traversaro, 2018-06-15:
We noted that robots-configuration is missing its dependency on ICUBcontrib .

traversaro, 2018-06-15:
First trial: there are some problem in finding the following devices: 
~~~
753	3,295896	ERROR	Cannot find "imuBosch_BNO055" plugin (not built in, and no .ini file found for it)
754	3,295930	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/imuBosch_BNO055.ini or share/yarp/plugins/imuBosch_BNO055.ini in it
756	3,295943	ERROR	cannot create device <imuBosch_BNO055>
757	3,295949	ERROR	ControlBoardWrapper: error while opening subdevice
777	3,569666	ERROR	Cannot find "embObjIMU" plugin (not built in, and no .ini file found for it)
778	3,569690	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/embObjIMU.ini or share/yarp/plugins/embObjIMU.ini in it
785	3,579674	ERROR	Cannot find "embObjIMU" plugin (not built in, and no .ini file found for it)
786	3,579701	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/embObjIMU.ini or share/yarp/plugins/embObjIMU.ini in it
792	3,589657	ERROR	Cannot find "embObjIMU" plugin (not built in, and no .ini file found for it)
793	3,589757	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/embObjIMU.ini or share/yarp/plugins/embObjIMU.ini in it
799	3,590014	ERROR	Cannot find "embObjIMU" plugin (not built in, and no .ini file found for it)
800	3,590052	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/embObjIMU.ini or share/yarp/plugins/embObjIMU.ini in it
823	3,751146	ERROR	Cannot find "skinWrapper" plugin (not built in, and no .ini file found for it)
824	3,751211	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/skinWrapper.ini or share/yarp/plugins/skinWrapper.ini in it
838	4,519874	ERROR	Cannot find "skinWrapper" plugin (not built in, and no .ini file found for it)
839	4,519892	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/skinWrapper.ini or share/yarp/plugins/skinWrapper.ini in it
852	5,279636	ERROR	Cannot find "skinWrapper" plugin (not built in, and no .ini file found for it)
853	5,279665	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/skinWrapper.ini or share/yarp/plugins/skinWrapper.ini in it
883	5,875075	ERROR	Cannot find "skinWrapper" plugin (not built in, and no .ini file found for it)
884	5,875100	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/skinWrapper.ini or share/yarp/plugins/skinWrapper.ini in it
914	6,469398	ERROR	Cannot find "skinWrapper" plugin (not built in, and no .ini file found for it)
915	6,469484	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/skinWrapper.ini or share/yarp/plugins/skinWrapper.ini in it
1003	9,246612	ERROR	Cannot find "embObjFTsensor" plugin (not built in, and no .ini file found for it)
1004	9,246645	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/embObjFTsensor.ini or share/yarp/plugins/embObjFTsensor.ini in it
1009	9,246696	ERROR	Cannot find "embObjFTsensor" plugin (not built in, and no .ini file found for it)
1010	9,246706	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/embObjFTsensor.ini or share/yarp/plugins/embObjFTsensor.ini in it
1015	9,256713	ERROR	Cannot find "embObjFTsensor" plugin (not built in, and no .ini file found for it)
1016	9,256755	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/embObjFTsensor.ini or share/yarp/plugins/embObjFTsensor.ini in it
1021	9,256830	ERROR	Cannot find "embObjFTsensor" plugin (not built in, and no .ini file found for it)
1022	9,256846	ERROR	Check that YARP_DATA_DIRS leads to at least one directory with plugins/embObjFTsensor.ini or share/yarp/plugins/embObjFTsensor.ini in it
~~~

traversaro, 2018-06-17:
The test was successful. With the fixes provided in https://github.com/robotology/robotology-superbuild/pull/90 , the superbuild is working correctly on the `icub-head`, and we were able to run the yoga++ demo on it. 

Current open points on the options: 
* I think  `ROBOTOLOGY_ENABLE_ICUB_ROBOT_ETH` and `ROBOTOLOGY_ENABLE_ICUB_ROBOT_CAN` are bad names. Someone (I think @prashanthr05) was already confused by them and thought they were necessary to use the iCub robot from an external PC. 
* To avoid confusion, I think it would make sense to just have a single option to enable to compile the device on the iCub's robot head. Perhaps this would mean some additional device compilation on some robots, but I think this is a reasonable price for the additional simplicity. However there are some devices (such as `cfw2can`) that can be compiled only if the CAN API package are present in the computer, so we should take care of them. 
Given this two points, I think it would make sense to have just two options: `ROBOTOLOGY_ENABLE_ICUB_HEAD` to compile all the software/devices necessary on the icub head, and an additional option `ROBOTOLOGY_ENABLE_ICUB_HEAD_CAN` to enable `cfw2can` and eventual other CAN-related devices. There are also the device `pcan` and `ecan` that are necessary on really old icub, be we should check if those are still supported (checking the `robots-configurations` repo, it does not seem that are still used).

traversaro, 2018-06-17:
Other doubt: 
* Should we include [`icub-firmware-build`](https://github.com/robotology/icub-firmware-build) and  [`icub-firmware`](https://github.com/robotology/icub-firmware) in the superbuild?

S-Dafarra, 2018-06-17:
What about robots-configuration? Before the configuration files were installed in the iCubContrib folder. Where should be installed now? Are they within the superbuild?

traversaro, 2018-06-17:
> What about robots-configuration? Before the configuration files were installed in the iCubContrib folder. Where should they be installed now? Are they within the superbuild?

`ICUBcontrib` is part of the robotology-superbuild, and it installs in the usual location of the superbuild `/usr/local/src/robot/src/robotology-superbuild/build/install`. `robots-configuration` finds the `ICUBcontrib` installed by the superbuild, and so it also installs in `/usr/local/src/robot/src/robotology-superbuild/build/install` . See https://github.com/robotology/robotology-superbuild/pull/79 for the discussion on adding `ICUBcontrib` to the superbuild. 

Note that as `ICUBcontrib` is part of the superbuild, also project external to the superbuild that use `ICUBcontrib` will install in the superbuild's install prefix, and I fear this could be a source of confusion. 

All the alias related to the `robots-configuration` contained in the `iCubGenova04`'s`.bashrc` have been updated.

traversaro, 2018-08-21:
In https://github.com/robotology/robotology-superbuild/issues/109 a fair point about the installation logic of `robots-configuration` is raised. We never realized that while using the superbuild on the robot because in https://git.robotology.eu/STraversaro/icub-environment/raw/use-robotology-superbuild/home/bashrc_iCub the `YARP_ROBOT_NAME` is always set to the values specified in the `yarp_robot_name.txt` `file.

traversaro, 2019-03-13:
Most of the issues discussed in here should be solved by https://github.com/robotology/robotology-superbuild/pull/183 .

traversaro, 2019-03-13:
bashrc file for the icub-head in https://git.robotology.eu/STraversaro/icub-environment/blob/use-robotology-superbuild/home/bashrc_iCub modified to remove all usage of `_ROOT` and `_DIR` variables, as discussed  in https://github.com/robotology/robotology-superbuild/issues/154 .

traversaro, 2019-03-13:
@gabrielenava if someone is testing the modifications on iCubGenova04, it would be nice to try also the new bashrc_iCub in https://git.robotology.eu/STraversaro/icub-environment/blob/use-robotology-superbuild/home/bashrc_iCub .

gabrielenava, 2019-03-13:
cc @GiulioRomualdi

traversaro, 2019-11-08:
I think that the superbuild is used successfuly on iCubGenova04 and iCubGenova01, and @valegagge , @vtikha  are currently experimenting with it for use for general robot deployment, so I think we can close this issue.

closed 2019-11-08 by traversaro