robotology/robotology-superbuild #92: Error related to the ICUBcontribConfig.cmake file

traversaro, 2018-06-19:
~~~
icub@icub-head:/usr/local/src/robot/robotology-superbuild/build$ make
-- YCM not found. Bootstrapping it.
-- Downloading YCMEPHelper.cmake - Attempt 1 of 3
-- Downloading YCMEPHelper.cmake - SUCCESS
-- Downloading StyleGITHUB.cmake - Attempt 1 of 3
-- Downloading StyleGITHUB.cmake - SUCCESS
-- Performing download step (git clone) for 'YCM'
-- Performing update step for 'YCM'
-- Performing configure step for 'YCM'
-- Performing uninstall step for 'YCM'
-- Performing build step for 'YCM'
-- Performing install step for 'YCM'
-- Found YCM: /usr/local/src/robot/robotology-superbuild/build/install/share/YCM (found version "0.5.1.20171220.1+gite9bfa39")
-- Package RTF not found. Will be downloaded and built.
-- Using iCub from install
-- Package iDynTree not found. Will be downloaded and built.
-- Package wholeBodyInterface not found. Will be downloaded and built.
-- Package yarpWholeBodyInterface not found. Will be downloaded and built.
-- Package codyco-modules not found. Will be downloaded and built.
CMake Deprecation Warning at build/install/lib/cmake/YARP/YARPConfig.cmake:272 (message):
  The YARP_MODULE_PATH variable is deprecated.  CMake find package modules
  are now in YCM.
Call Stack (most recent call first):
  build/install/lib/ICUBcontrib/ICUBcontribConfig.cmake:9999 (_yarp_module_path_is_deprecated)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:180 (find_package)
  cmake/Buildrobots-configuration.cmake:9 (find_or_build_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:235 (include)
  CMakeLists.txt:152 (find_or_build_package)

CMake Error at build/install/lib/ICUBcontrib/ICUBcontribConfig.cmake:12 (include):
  include could not find load file:

    YarpInstallationHelpers
Call Stack (most recent call first):
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:180 (find_package)
  cmake/Buildrobots-configuration.cmake:9 (find_or_build_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:235 (include)
  CMakeLists.txt:152 (find_or_build_package)

CMake Deprecation Warning at build/install/lib/cmake/YARP/YARPConfig.cmake:272 (message):
  The YARP_MODULE_PATH variable is deprecated.  CMake find package modules
  are now in YCM.
Call Stack (most recent call first):
  build/install/lib/ICUBcontrib/ICUBcontribConfig.cmake:9999 (_yarp_module_path_is_deprecated)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:287 (find_package)
  cmake/Buildrobots-configuration.cmake:9 (find_or_build_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:235 (include)
  CMakeLists.txt:152 (find_or_build_package)

CMake Error at build/install/lib/ICUBcontrib/ICUBcontribConfig.cmake:12 (include):
  include could not find load file:

    YarpInstallationHelpers
Call Stack (most recent call first):
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:287 (find_package)
  cmake/Buildrobots-configuration.cmake:9 (find_or_build_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:235 (include)
  CMakeLists.txt:152 (find_or_build_package)

-- Package robots-configuration not found. Will be downloaded and built.
-- 
-- The following RUNTIME packages have been found:

 * Git

-- The following OPTIONAL packages have been found:

 * ICUB
 * ICUBcontrib

-- The following REQUIRED packages have been found:

 * YARP_dev
 * YARP_conf
 * YARP_OS
 * YARP_sig
 * YARP_math
 * YARP

-- The following OPTIONAL packages have not been found:

 * RTF
 * iDynTree
 * codyco-modules
 * wholeBodyInterface
 * yarpWholeBodyInterface
 * robots-configuration

-- The following RECOMMENDED packages have not been found:

 * YCM (required version >= 0.8.1) , A collection of CMake modules and utilities for YARP and friends , <https://github.com/robotology/ycm>
   Used by the build system

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/robot/robotology-superbuild/build/CMakeFiles/CMakeOutput.log".
Makefile:1222: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
~~~

traversaro, 2018-06-19:
This strange behaviour, where the superbuild seems to be finding its own projects, was probably due to a dirty system, that contained a `CMakeFiles` and a `CMakeCache.txt` file contained in the root of the superbuild. I am unable to reproduce it on my computer, but for the meantime I think we should close this issue.

traversaro, 2018-06-26:
The issue appear again, apparently without any kind of in-source build creating problems. 
@drdanz do you know which part of the code is preventing the superbuild to find its own subprojects even if they are listed in the `CMAKE_PREFIX_PATH` enviromental variable?

traversaro, 2018-06-26:
The specific error related to the `ICUBcontrib` was fixed in https://github.com/robotology/icub-contrib-common/commit/a4e4cb8f1961277454dbca385cebdd17384bbcc0    , but that was only a byproduct of the actual issue of the superbuild that finds its own subproject components on Ubuntu 16.04 . 

I am unable to reproduce the problem even enabling `ROBOTOLOGY_ENABLE_ICUB_ROBOT_ETH` on a Ubuntu 18.04 system with CMake 3.10 . I wonder  if the issue was related to https://github.com/robotology/yarp/issues/1741 , that was recently merged.

traversaro, 2018-06-26:
The problem is still present:
~~~
-- YCM not found. Bootstrapping it.
-- Downloading YCMEPHelper.cmake - Attempt 1 of 3
-- Downloading YCMEPHelper.cmake - SUCCESS
-- Downloading StyleGITHUB.cmake - Attempt 1 of 3
-- Downloading StyleGITHUB.cmake - SUCCESS
-- Performing download step (git clone) for 'YCM'
-- Performing update step for 'YCM'
-- Performing configure step for 'YCM'
-- Performing uninstall step for 'YCM'
-- Performing build step for 'YCM'
-- Performing install step for 'YCM'
-- Found YCM: /usr/local/src/robot/robotology-superbuild/build/install/share/YCM (found version "0.8.1")
-- Package RTF not found. Will be downloaded and built.
-- Using iCub from install
-- Package icub-tests not found. Will be downloaded and built.
-- Package iDynTree not found. Will be downloaded and built.
-- Package wholeBodyInterface not found. Will be downloaded and built.
-- Package yarpWholeBodyInterface not found. Will be downloaded and built.
-- Package codyco-modules not found. Will be downloaded and built.
-- Package robots-configuration not found. Will be downloaded and built.
-- 
-- The following RUNTIME packages have been found:

 * Git

-- The following OPTIONAL packages have been found:

 * ICUB
 * YARP_dev
 * YARP_conf
 * YARP_OS
 * YARP_sig
 * YARP_math
 * ICUBcontrib

-- The following REQUIRED packages have been found:

 * YARP , A library and toolkit for communication and device interfaces , <https://www.yarp.it/>

-- The following OPTIONAL packages have not been found:

 * RTF
 * icub-tests
 * iDynTree
 * codyco-modules
 * wholeBodyInterface
 * yarpWholeBodyInterface
 * robots-configuration

-- The following RECOMMENDED packages have not been found:

 * YCM (required version >= 0.8.1) , A collection of CMake modules and utilities for YARP and friends , <https://github.com/robotology/ycm>
   Used by the build system

-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/src/robot/robotology-superbuild/build
~~~

drdanz, 2018-07-02:
> @drdanz do you know which part of the code is preventing the superbuild to find its own subprojects even if they are listed in the CMAKE_PREFIX_PATH enviromental variable?

I don't remember if there was a way to avoid that...

traversaro, 2019-01-22:
This happened again on Ubuntu 18.04 on the head of the iCubGenova04. 
cc @GiulioRomualdi @kouroshD @S-Dafarra

traversaro, 2019-01-22:
We found the problem (at least I think, @GiulioRomualdi @kouroshD correct me if I am wrong): if the superbuild install prefix is in the `CMAKE_PREFIX_PATH` enviromental variable, the superbuild ignores it correctly (given @drdanz answer, I do not know if this is the intended behavior, but is the one that we have used in years with the superbuild). If instead the `<package>_DIR` env variable is set  for some `<package>` of the superbuild, the superbuild does not ignore it, and so `find_or_build_package` is able to find its own packages.

traversaro, 2019-12-09:
> 
> 
> > @drdanz do you know which part of the code is preventing the superbuild to find its own subprojects even if they are listed in the CMAKE_PREFIX_PATH enviromental variable?
> 
> I don't remember if there was a way to avoid that...

I am afraid that the part of code that prevents that is https://github.com/robotology/ycm/blob/7cc2b755e302b972112ca6b77a6dbce4492d0062/tools/YCMBootstrap.cmake#L75 , and so the projects in the superbuild on a second build of the superbuild are correctly ignored only if the YCM is bootstrapped, and not if an external YCM is used (I wonder how we never noticed that :/).

traversaro, 2019-12-09:
Related commit and  issue: 
* https://github.com/robotology/ycm/commit/20980db43457e148bb5200a02da7ada6566a1b0f
*  https://github.com/robotology/ycm/issues/21

traversaro, 2020-01-30:
A possible way to mitigate this problem is to set `USE_SYSTEM_<package>` CMake variable to force the use of superbuild for that package, see http://robotology.github.io/ycm/gh-pages/git-master/manual/ycm-superbuild.7.html#ycm-superbuild-manual-for-developers. I think the best way to mitigate this is to fix is to actually have some variable `YCM_USE_SYSTEM_ALL` that forces all the projects contained in the superbuild to ignore the system version. cc @HosameldinMohamed

HosameldinMohamed, 2020-01-30:
> A possible way to mitigate this problem is to set USE_SYSTEM_<package> CMake variable to force the use of superbuild for that package

Thanks @traversaro! I had the target `YARP` missing for example and once I set the variable `USE_SYSTEM_YARP` (found in the advanced CMake options) to FALSE, I was able to build the package.

traversaro, 2020-04-17:
If you use `Unstable` branches in the robotology-superbuild, there is now available the `YCM_DISABLE_SYSTEM_PACKAGES` CMake cache option (see https://github.com/robotology/ycm/pull/332) that can be set to `ON` to automatically avoid finding the projects installed in the system if <package>_DIR env variables are defined . Once a new feature release of YCM is done, the idea is to set this option to `ON` by default, and finally close this issue.

traversaro, 2021-09-07:
> 
> 
> If you use `Unstable` branches in the robotology-superbuild, there is now available the `YCM_DISABLE_SYSTEM_PACKAGES` CMake cache option (see [robotology/ycm#332](https://github.com/robotology/ycm/pull/332)) that can be set to `ON` to automatically avoid finding the projects installed in the system if _DIR env variables are defined . Once a new feature release of YCM is done, the idea is to set this option to `ON` by default, and finally close this issue.

Done in https://github.com/robotology/robotology-superbuild/pull/389 , and the features was release in YCM 0.12 that was released on 28 Sep 2020 . I think we can close the issue, and open it again if necessary.

closed 2021-09-07 by traversaro