robotology/robotology-superbuild #95: Add `<prefix>/lib/rtf` to LD_LIBRARY_PATH ?

traversaro, 2018-06-22:
Otherwise, I don't think RTF plugins for YARP are working, see https://github.com/robotology/yarp/issues/1375 .

drdanz, 2018-07-02:
This is a workaround. I'd rather fix RTF instead...

drdanz, 2018-07-05:
Setting `LD_LIBRARY_PATH` is usually considered a bad practice...
I'd like to standardize some mechanism in shlibpp, but I don't know exactly how yet

diegoferigo, 2018-07-05:
I'm interested in this, but I have no idea on how to achieve it without `LD_LIBRARY_PATH`.

traversaro, 2018-07-05:
I checked a few "`LD_LIBRARY_PATH` is bad" articles: 
* http://xahlee.info/UnixResource_dir/_/ldpath.html
* https://stackoverflow.com/questions/25918462/why-ld-library-path-is-bad-and-the-correct-way-to-load-dynamic-libraries 
* https://gms.tf/ld_library_path-considered-harmful.html#patch-existing-elf-binaries
but none of them mention `dlopen` or dynamic plugin loading.

claudiofantacci, 2018-07-06:
> I'm interested in this, but I have no idea on how to achieve it without LD_LIBRARY_PATH.

Me too! Just came through this "problem" as well while setting up the VM for EPR. I don't really like putting things in `LD_LIBRARY_PATH`.

drdanz, 2018-07-13:
> I'm interested in this, but I have no idea on how to achieve it without `LD_LIBRARY_PATH`.

The idea should be to have some configuration or some manifest file to locate the library, and then open it using the full path.

Unfortunately there are quite a lot of issues with this:

0. Adding further environment variables is probably not such a good idea.
1. If you write a manifest file, it should not contain absolute paths in order to make the package re-locatable
2. multi config generators (i.e. MSVC, Xcode) put the libraries in subfolders
3. on windows, usually, the dll has a postfix `d` in the name
4. every system uses custom mechanisms and default paths (just to make a couple of examples, `ldconfig` is used on linux, the `PATH` variable is searched on windows)

This is what [YARP](https://github.com/robotology/yarp/blob/master/src/libYARP_OS/src/YarpPlugin.cpp)  does, just to make an example.

traversaro, 2018-07-18:
I think we should add it until we found a better solution.

traversaro, 2018-07-18:
This is what YARP documentation suggests after all https://github.com/robotology/yarp/blame/v3.0.0/doc/rtf_plugins.md#L14 .

diegoferigo, 2018-07-18:
Waiting a smarter solution, which will take time and brain power, I would add it as well.

traversaro, 2019-01-22:
Several projects define their own `*_PLUGIN_PATH` environmental variable:
*  `QT_PLUGIN_PATH` : http://doc.qt.io/qt-5/deployment-plugins.html 
* `GAZEBO_PLUGIN_PATH` : http://gazebosim.org/tutorials/?tut=plugins_hello_world
* `IGN_PLUGIN_PATH` : https://ignitionrobotics.org/api/common/3.0/classignition_1_1common_1_1SystemPaths.html#a7b4501e2e593c5e088e33bbbf1e3adbf

It would be nice if there could be a consensus among these kind of projects to support a `PLUGIN_PATH` environmental variable such that it could be just set once for each "non-global" prefix.
 
However, sticking with the robotology-superbuild problem, we have this problem for two projects: 
* `RTF` 
* `BlockFactory` 

A possible solution would be if these two projects defined their own `RTF_PLUGIN_PATH` and `BLOCKFACTORY_PLUGIN_PATH` environmental variables. Even if the projects use different environmental variables, it would probably make sense to add a [`FindSharedLibrary`](https://ignitionrobotics.org/api/common/3.0/classignition_1_1common_1_1SystemPaths.html#a7b4501e2e593c5e088e33bbbf1e3adbf)-like to  [`sharedlibpp`](https://github.com/robotology-playground/sharedlibpp) to find an arbitrary plugin given a established search path (that could be customized by specifying the `<project>_PLUGIN_PATH` env variable name. 

If instead we want to try to establish a consensus on a shared environmental variable, we could make it easy to use `PLUGIN_PATH` for [`sharedlibpp`](https://github.com/robotology-playground/sharedlibpp), hoping that it is not conflicting with any existing env variable. : )

diegoferigo, 2019-01-23:
I had in mind exactly the approach used in ignition. I found it great also for what concern the tests, where the plugin folder inside the build directory was added in the search path automatically, without the need to manually operate on `LD_LIBRARY_PATH`.

Despite `PLUGIN_PATH` is appealing, for the time being, at least for what concern BlockFactory, I would proceed with a `BLOCKFACTORY_PLUGIN_PATH` env variable, maybe matched with custom cmake commands for installing the plugin library in the right folder.

traversaro, 2019-01-23:
> Despite `PLUGIN_PATH` is appealing, for the time being, at least for what concern BlockFactory, I would proceed with a `BLOCKFACTORY_PLUGIN_PATH` env variable, maybe matched with custom cmake commands for installing the plugin library in the right folder.

I totally agree. Do you think anyhow the logic for finding the library can be added to the `sharedlibpp` ? This may be a quite useful to easily generalize the approach to RTF.

diegoferigo, 2019-01-31:
In https://github.com/robotology/blockfactory/pull/32 I edited the `shlibpp` version vendored by blockfactory to support the following:

- A generic `SHLIBPP_PLUGIN_PATH` env variable to specify extra plugin paths
- Extending the plugin search path by calling a new method

traversaro, 2019-02-03:
For reference, apparently in ROS's world this "finding the plugin path" is handled by pluginlib ( https://github.com/ros/pluginlib ) while most of the logic that we have  in `shlibpp`. 
Interestingly, the directory for searching for plugins are found by append `lib` (or `bin` in ROS2 and the latest ROS from Microsoft) to some sort of prefi. For ROS1, the `CMAKE_PREFIX_PATH` env variable is used, while in ROS2 the `get_package_prefix`'s ament_cpp function.  See the relevant code: 
* ROS1 : https://github.com/ros/pluginlib/blob/0a3c0de442596b46026eb4c738130bce90421927/pluginlib/include/pluginlib/class_loader_imp.hpp#L302
* ROS2 : https://github.com/ros/pluginlib/blob/c2a604b220803a8fc22168935278e7b0e3d13a10/pluginlib/include/pluginlib/class_loader_imp.hpp#L397

diegoferigo, 2019-02-04:
Using `CMAKE_PREFIX_PATH` is actually a good choice, at least for installed projects. In my opinion tho having a `*_PLUGIN_PATH` variable can be useful for developers whenever you want to use the build tree, just to make an example.

traversaro, 2019-02-04:
> Using `CMAKE_PREFIX_PATH` is actually a good choice, at least for installed projects. 

I think it really depends on **how** the project is installed. If the plugin is installed by the `.deb` package in the `/usr` prefix for execution in a system that is not used for development, the need for setting `CMAKE_PREFIX_PATH` (in a system that may not have even `CMake` installed) seems a bit odd. I guess it could make sense to always (i.e. hardcode) the search in the relevant sub-directories of the path in [`CMAKE_SYSTEM_PREFIX_PATH`](https://cmake.org/cmake/help/v3.13/variable/CMAKE_SYSTEM_PREFIX_PATH.html) (see https://github.com/Kitware/CMake/blob/0479ae492a6bc6058ae86d3ce01ac26fcc71c744/Modules/Platform/UnixPaths.cmake#L24 for the definition in *nix system).

traversaro, 2019-02-04:
Great, I did not realized this was already supported by https://github.com/robotology/blockfactory/pull/32/files#diff-ca9447c1aff58642a23f971df7d32a95R138 .

drdanz, 2019-02-13:
Can we try to synchronize the versions of `shlibpp` that we have around? AFAIK there are 5 at the moment:

* robotology/shlibpp
* robotology/yarp
* robotology/robot-testing
* robotology/wbtoolbox
* robotology/block-factory

diegoferigo, 2019-02-13:
@drdanz You can strikeout `wb-toolbox` since from the v5 version (current `master`) it uses the version vendored by blockfactory. Did you have a look by any chance to the improvements of https://github.com/robotology/blockfactory/pull/32?

traversaro, 2019-03-13:
I think the changes in https://github.com/robotology/blockfactory/pull/32 are good to be submitted to https://github.com/robotology-playground/sharedlibpp .

diegoferigo, 2019-03-13:
Regardless of the possibility to make `shlibpp` a dependency of yarp / rtf / etc, merging those features upstream would be a nice addition. Once upstream is aligned, we can add `shlibpp` to the superbuild, initially as `blockfactory` dependency.

cc @S-Dafarra is interested in this merge too

drdanz, 2019-03-14:
> Did you have a look by any chance to the improvements of robotology/blockfactory#32?

Sorry, I didn't have time yet... Maybe you could open a PR on shlibpp and we can discuss the changes there?

traversaro, 2020-04-26:
cc @Nicogene as he may be interested in all the shlibpp-related discussion.

traversaro, 2020-04-26:
A fix for this issue is provided in https://github.com/robotology/robotology-superbuild/pull/393 . A "proper" fix for RTF will be tracked in https://github.com/robotology/robot-testing-framework/issues/109 .

closed 2020-04-26 by traversaro