robotology/robotology-superbuild #101: Add a CI job building projects in devel branch

traversaro, 2018-07-26:
This can be done with https://github.com/robotology/robotology-superbuild/pull/100 . 
Related issue: https://github.com/robotology/robotology-superbuild/issues/96 .

traversaro, 2019-09-27:
>  Related issue: #96 .

Things has changed, I think now it makes more sense to set it up with GitHub Actions.

diegoferigo, 2019-10-15:
If the tests will be executed in a clean docker image by building a given Dockerfile, in order to test both devel and master with the same file you can refer to:

https://github.com/robotology-playground/docker_images/blob/cb4ab232a327b03ae4c84fd9e3d2f0bb8a6fc213/robotology-tdd/Dockerfile#L71-L88

Bear in mind that this approach does not support ccache. Especially for big projects like this one, ccache support is a must-have.

traversaro, 2019-10-15:
My current plan is to actually add a CMake option to switch between *stable* (the old "master") and *unstable* (the old "devel") tags, documenting clearly that switching after the first project generation or having multiple builds with different settings is not supported at all. This is not ideal solution, but I think better then the status-quo. Once that is in place, I think it make sense to have GitHub Actions jobs dedicated to build them.

diegoferigo, 2019-10-15:
Even better :+1: I would add also the support to load a custom `ProjectTags*.cmake`.

traversaro, 2019-10-15:
Good idea.

nunoguedelha, 2019-10-15:
Let me know if you guys need any help with that.

closed 2019-11-07 by traversaro