robotology/robotology-superbuild #109: Issues in running iKinGazeCtrl after switching to superbuild

serena-ivaldi, 2018-08-21:
Hello, we recently switched to the superbuild, and now we are having problems in launching even the most basic demos, as it seems that the configuration files are misplaced.
For example, when we launch iKinGazeCtrl:
`iKinGazeCtrl --from iKinGazeCtrl.ini`
it doesn't find the ini file. It can be a problem of our YARP_DATA_DIRS, but we followed the instructions in the doc. So my question (that helps me debug) is where is this file supposed to be?
Because in robotology-superbuild/build/install/share/iCub/contexts/iKinGazeCtrl we have the context for the gazeCtrl but not this file, that I find in the sources of iCubNancy01 in /robotology-superbuild/robotology/robots-configuration ... but apparently these are not installed in share... did we make a mistake? I don't find a way to install the files of our iCubNancy01 robot, there is not a specific field in the cmake.
Side problem is that now when we launch our robot, he is called /iCubNancy01/ as prefix in all ports, and I cannot find a way to specify this new prefix in iKinGazeCtrl (because I cannot trace where it is looking for the ini files).

Incidentally, when launching iKinGazeCtrl I can see it is looking for two files:
yarp: cannot read from iKinGazeCtrl.ini
yarp: cannot read from tweak.ini
but we don't have a file called tweak.ini

Any help @pattacini on the location of the config files inside the superbuild? 
Any help in figuring out what we did wrong in the installation @traversaro ?

traversaro, 2018-08-21:
Ok, `INSTALL_ALL_ROBOTS` is currently broken: https://github.com/robotology/robots-configuration/issues/77 .

traversaro, 2018-08-21:
Another alternative is to expose the `ROBOT_NAME` option as an option of the robotology-superbuild, but I am not a big fan of this option.

traversaro, 2018-08-23:
@serena-ivaldi If you try it, let us know if the suggest workaround is working, thanks.

traversaro, 2018-09-12:
@InriaBrice any update on this?

InriaBrice, 2018-09-12:
Hey @traversaro , sorry for the delay. I just checked and i added to YARP_DATA_DIRS the following path : 
~/software/src/robots-configuration/iCubNancy01

So now when i do iKinGazeCtrl --from iKinGazeCtrl.ini he can find it.
To be honest i don't really like the idea of the installation because it's a bit confusing to have a file installed which is used and another which should be edited. It's a common thing to update original file and forgot to install. Anyway,

For the iKinGazeCtrl stuff, in iCubStartup.xml we had wrong robot name, it was looking for /iCubNancy01/yarprobotinterface but /icub/yarprobotinterface was written.

I changed also head-inertial.xml param name /icub/inerial:o to /iCubNancy01/inertial:o so imuFilter connection can find it too.

In all our config files, there is no robot_name variable (in robots-configuration) so i don't think that changing robot_name cmake variable will do anything. BUT it was well set to iCubNancy01.

Best regards,

traversaro, 2018-09-12:
>  I just checked and i added to YARP_DATA_DIRS the following path :
~/software/src/robots-configuration/iCubNancy01

Ok, so you are not managing anymore the `robots-configuration` repo with the superbuild, contrary to what I understood from https://github.com/robotology/robotology-superbuild/issues/109#issue-352609210 , right? 
I think this is a good idea, because as clearly stated in https://github.com/robotology/robotology-superbuild/issues/88 , the usage of the robot-superbuild inside the robot's setup is still experimental and not recommended for users outside of IIT.

> To be honest i don't really like the idea of the installation because it's a bit confusing to have a file installed which is used and another which should be edited. It's a common thing to update original file and forgot to install. Anyway,

I totally understand, this is a common problem, and the reason build system such as `bazel` do not actually support any installation out of the box. I typically deal with that by always using `make install` (or in the case of the superbuild, just use `make` as this will `make install` all the subprojects). However, the superbuild is just a simplified way of managing the dependencies, and for this reason we document/support in the superbuild the most elementary way of handing the configuration files. As long as you have full control of your setup, you can arrange the projects and use the configuration files in the configuration that is more suitable for you, as long as  you follow http://www.yarp.it/yarp_data_dirs.html . 

> In all our config files, there is no robot_name variable (in robots-configuration) so i don't think that changing robot_name cmake variable will do anything. BUT it was well set to iCubNancy01.

This is strange. In `robots-configuration`'s CMake files the `ROBOT_NAME` is used in https://github.com/robotology/robots-configuration/blob/master/CMakeLists.txt#L19 . If `ROBOT_NAME` is correctly set to `iCubNancy01` and you install the project by using `<prefix>` as `CMAKE_INSTALL_PREFIX`, it should be sufficient to add `<prefix>/share/iCub` to `YARP_DATA_DIRS` and set the `YARP_ROBOT_NAME` enviromental variable to `iCubNancy01` to find all the configuration files, without the need to add the specific  iCubNancy01's source  directory `YARP_DATA_DIRS` .

> For the iKinGazeCtrl stuff, in iCubStartup.xml we had wrong robot name, it was looking for /iCubNancy01/yarprobotinterface but /icub/yarprobotinterface was written.
>
> I changed also head-inertial.xml param name /icub/inerial:o to /iCubNancy01/inertial:o so imuFilter connection can find it too.

This is strange, and I think I cannot help you here. If you need any modifications to the configuration files in `robots-configuration` to get your robot to work, I suggest you to submit a PR with the necessary modifications to https://github.com/robotology/robots-configuration , so if those modifications are actually necessary they will be merged, or otherwise the people managing the configuration files can check what is going on, thanks!

InriaBrice, 2018-09-12:
Just one question, in 

```
<module>
                <name>imuFilter</name>
                <node>pc104</node>
                <dependencies>
                    <port timeout="60">/imuFilter/inertial:i</port>
                </dependencies>
        </module>

        <connection>
                <from>/iCubNancy01/inertial:</from>
                <to>/imuFilter/inertial:i</to>
                <protocol>tcp</protocol>
        </connection>
```
the dependencies port are created or should already been created ?

I can't find the documentation about theses xml variables.

traversaro, 2018-09-12:
Can you post a link to that file? 
See http://www.yarp.it/yarpmanager.html#application for the documentation on the `<dependencies>` tag.  In a nutshell " module execution will be postponed until the dependencies are satisfied", so those ports should be open **before** the manager starts to launch the module.

InriaBrice, 2018-09-12:
It's a configuration file that we have, and i'm trying to understand. But i guess i got my answer : as you said <dependies> </dependencies> are the things needed to run the module.

Thanks for the documentation it was what i was looking for !

traversaro, 2018-09-12:
> It's a configuration file that we have, and i'm trying to understand. 

Ok, in that case it seems wrong, as I would say that the `/imuFilter/inertial:i` port should be opened by the `imuFilter` module, so it does not make sense that the module waits on it.

closed 2019-03-15 by traversaro