robotology/robotology-superbuild #111: Gazebo models installed by icub-models are not available in Gazebo after installing the robotology-superbuild

traversaro, 2018-08-23:
I just realized discussing with @pattacini that the Gazebo models installed by `icub-models` are not available in Gazebo, because in the superbuild docs and setup.sh file we do not add to `GAZEBO_MODEL_PATH` the  directories `<prefix>/share/iCub` and `<prefix>`, as documented in https://github.com/robotology/icub-models#use-the-models-with-gazebo . 
To avoid confusion to superbuild users, perhaps we  should first solve  https://github.com/robotology/icub-models/issues/11 ? 
@diegoferigo @gabrielenava

diegoferigo, 2018-08-24:
> To avoid confusion to superbuild users, perhaps we should first solve robotology/icub-models#11 ?

Yes, I would proceed in this direction

diegoferigo, 2018-08-24:
cc @nunoguedelha

traversaro, 2018-08-24:
> I am not a user of setup.sh since I prefer to manually set the paths in the .bashrc, and I didn't note that the setup.sh was not up to date

I see, but consider that also the docs to write your own `setup.sh` are missing that part.

gabrielenava, 2018-10-22:
Given that https://github.com/robotology/icub-models#use-the-models-with-gazebo is now closed, I may take this issue and fix the setup.sh if there are no other reasons for waiting.

traversaro, 2018-10-22:
I do not think anything else is blocking the issue.

gabrielenava, 2018-10-23:
Fixed with this commit: https://github.com/robotology/robotology-superbuild/commit/32c56bc2d390023f5e16dc09776039f648fa1067

Closing the issue.

closed 2018-10-23 by gabrielenava