robotology/robotology-superbuild #115: Reduce the number of Travis jobs

traversaro, 2018-09-12:
8 jobs on Linux are totally useless. 

I would like to reduce them, and just have a limited number of jobs out of the permutations of [(gcc|clang), (Debug|Release), (bionic|xenial)], so we can eventually add other distros (I think debian sid in particular is very useful to catch early any kind of problem due to updated Debian dependencies). 

@diegoferigo Any idea on the correct way to write the job matrix? I tried but in the end it was never working, either all the jobs were starting anyway, or no job was starting at all.

diegoferigo, 2018-09-12:
Strange, these lines define the build matrix for the `test` stage:

https://github.com/robotology/robotology-superbuild/blob/f418c827be5e20ccb4314335aed3e15fc2bbc893/.travis.yml#L16-L27

Plus, there are two jobs for osx:

https://github.com/robotology/robotology-superbuild/blob/f418c827be5e20ccb4314335aed3e15fc2bbc893/.travis.yml#L76-L94

I think that removing / adding lines inside the `env:matrix` entry should be enough. Afterwards, they are doubled by the compiler entry.

diegoferigo, 2018-09-12:
>>  or no job was starting at all

Do you mean that Travis was not starting? In this case you might have a wrongly formatted yaml. I usually double check it before pushing using an online resource such as https://yaml-online-parser.appspot.com/.

traversaro, 2018-09-12:
Now I remember: I think I was trying to include the compiler information in the matrix line as I used to do in the non-docker version, but I guess that was creating problems.

diegoferigo, 2018-09-12:
If I recall you can still do it if you use the `jobs:include` entry, but it has the drawback that you need to specify one by one all the configuration.

traversaro, 2019-11-05:
This was probably fixed by https://github.com/robotology/robotology-superbuild/pull/197 . However, today I also deactivated Travis as it was definitely broken (walking-controllers require a newer Eigen version then the 3.3beta2 available in Xenial) and GitHub Actions is going to replace it, so I think we can close the issue.

closed 2019-11-05 by traversaro