robotology/robotology-superbuild #121: CMake error when trying to compile the superbuild with Yarp

gabrielenava, 2018-09-24:
I downloaded from scratch the `robotology-superbuild` on my laptop and I tried to compile it. I do not have `Yarp` installed on my lapotop, it is supposed to be installed together with the superbuild. Nevertheless, when I tried to compile the project I got the following error:

```CMake

 CMake Error at /home/gnava/Software/github/INSTALL/lib/ICUBcontrib/ICUBcontribConfig.cmake:7 (find_package):
   By not providing "FindYARP.cmake" in CMAKE_MODULE_PATH this project has
   asked CMake to find a package configuration file provided by "YARP", but
   CMake did not find one.

   Could not find a package configuration file provided by "YARP" with any of
   the following names:

     YARPConfig.cmake
     yarp-config.cmake

   Add the installation prefix of "YARP" to CMAKE_PREFIX_PATH or set
   "YARP_DIR" to a directory containing one of the above files.  If "YARP"
   provides a separate development package or SDK, be sure it has been
   installed.
 Call Stack (most recent call first):
   build/install/share/YCM/modules/FindOrBuildPackage.cmake:180 (find_package)
   CMakeLists.txt:111 (find_or_build_package)

```

Maybe some path got broken recently?

##### I have Ubuntu 16.04 as OS

traversaro, 2018-09-24:
I guess you have an external installation of ICUBcontrib install as `CMAKE_INSTALL_PREFIX` :  `/home/gnava/Software/github/INSTALL/`, that was configured with a YARP installation that is not anymore installed or not anymore present in the `CMAKE_PREFIX_PATH`.

gabrielenava, 2018-09-24:
>I guess you have an external installation of ICUBcontrib install as CMAKE_INSTALL_PREFIX : /home/gnava/Software/github/INSTALL/, that was configured with a YARP installation that is not anymore installed or not anymore present in the CMAKE_PREFIX_PATH.

Yes, I have it. Ok then I will address this on my pc thanks!

traversaro, 2018-09-24:
Exactly, ICUBcontrib is now a project handled in the superbuild, so you either: 
* Install it with the superbuild 
or 
* use it externally, but all its dependencies should also be installed externally (in this case, YARP)