robotology/robotology-superbuild #124: Add funny-things, icub-basic-demos and speech repository

traversaro, 2018-09-27:
A recurring problem in setting up the robot for demonstration is the installation and configuration of the following repositories: 
* https://github.com/robotology/funny-things
* https://github.com/robotology/speech 
* https://github.com/robotology/icub-basic-demos

`icub-basic-demos` could go in the CORE profile, but for the other repos it could make sense to create a new profile. Any idea about the name? 

cc @gabrielenava @S-Dafarra @hu-yue

S-Dafarra, 2018-09-27:
What about ``BASIC_DEMOS`` profile for all of them?
Also the [``svox-speech``](https://github.com/robotology/speech/tree/master/svox-speech) module should be compiled, but it is inside the ``speech`` folder.

traversaro, 2018-09-27:
> What about ``BASIC_DEMOS`` profile for all of them?

I prefer `ICUB_BASIC_DEMOS`, but I like the idea. 

> Also the [``svox-speech``](https://github.com/robotology/speech/tree/master/svox-speech) module should be compiled, but it is inside the ``speech`` folder.

Speech issue opened: https://github.com/robotology/speech/issues/19 .

traversaro, 2019-03-12:
cc @pattacini @vtikha @randaz81 this is related to our discussion about the profiles.

closed 2020-02-17 by traversaro