robotology/robotology-superbuild #127: If `ROBOTOLOGY_ENABLE_ICUB_ROBOT_ETH` is enabled after a superbuild is compiled, the icub-firmware-shared dependency repo is not download

traversaro, 2018-10-11:
This issue should be addressed by https://github.com/robotology/robotology-superbuild/pull/126 , but that is just a workaround.

S-Dafarra, 2018-10-11:
Also, it would be better to enable the ``ICUB`` options accordingly. I followed the instructions in http://wiki.icub.org/wiki/Compilation_on_the_pc104, but I am not sure whether the list presented in there is complete.

S-Dafarra, 2018-10-11:
Sorry, I did not notice https://github.com/robotology/robotology-superbuild/blob/master/cmake/BuildICUB.cmake#L33-L57.

traversaro, 2018-10-11:
Not sure what you mean by "it falls in the same issue", but if I understood the problem it can be avoided by making all those options `CMAKE_ARGS` instead of `CMAKE_CACHE_ARGS`, see https://github.com/robotology/robotology-superbuild/issues/68 .

S-Dafarra, 2018-10-11:
> Not sure what you mean by "it falls in the same issue"
If `ROBOTOLOGY_ENABLE_ICUB_ROBOT_ETH` is enabled after a superbuild is compiled, those options are not set.

nunoguedelha, 2019-01-21:
I tried to reproduce this issue on an arbitrary machine (iCub head was not available) and the issue was not reproducible, even after removing the workaround implemented in #126.
So, when enabling the `ROBOTOLOGY_ENABLE_ICUB_ROBOT_ETH` option after a superbuild is compiled without it, the option is set in the cache and the repo `icub-firmware-shared` is downloaded.

Trying it now in the **icub04** head.

nunoguedelha, 2019-01-21:
As far as I understood from the discussions, and from [the link they mention](https://gitlab.kitware.com/cmake/cmake/commit/98cdb658e481a23c48ad894bec102ec5f060d72e?view=parallel), the proper keyword to use in `BuildICUB.cmake` for setting the `ROBOTOLOGY_ENABLE_ICUB_XXX` options, is `CMAKE_CACHE_DEFAULT_ARGS` rather than `CMAKE_CACHE_ARGS`:
https://github.com/robotology/robotology-superbuild/blob/51c1ba99e90d62013aa23901ede01758258a0040/cmake/BuildICUB.cmake#L33-L57

traversaro, 2019-01-21:
I am afraid that `CMAKE_CACHE_DEFAULT_ARGS` will mean that the option is set only the first time that ICUB is configured, so if a user configure first the superbuild with `ROBOTOLOGY_ENABLE_ICUB_ROBOT_ETH` set to `OFF`, and then with `ROBOTOLOGY_ENABLE_ICUB_ROBOT_ETH` set to `ON`, `iCub` will not be updated to reflect it.

nunoguedelha, 2019-01-21:
ok, then the documentation on this flag was a bit misleading, but it's clear now after our discussion one to one. I don't know if @S-Dafarra is aware of these details, so I will recap here what I understood from our discussion:
- `CMAKE_CACHE_DEFAULT_ARGS` allows the user to change the options that have been initialized in the CMake cache, under the sub-project (for instance running `ccmake .` in `<robotology-superbuild>/build/robotology/icub-firmware-shared/`), but afterwards cannot be overwritten by the superbuild CMake commands.
- `CMAKE_CACHE_ARGS` doesn't allow the user to change the options that have been initialized in the CMake cache under a sub-project, but he can still do it under the superbuild project through `ccmake .`  in `<robotology-superbuild>/build`, and it can be overwritten by the superbuild CMake commands if required.

nunoguedelha, 2019-01-21:
So bottom line: the current implementation should work perfectly fine, and we don't know what went wrong when @S-Dafarra observed the issue.

Anyway, I tried to reproduce the issue on the iCub04 head:
- cloned the `robotology-superbuild` into an empty folder,
- built everything from scratch with `ROBOTOLOGY_ENABLE_ICUB_ROBOT_ETH` set to `OFF`,
- set `ROBOTOLOGY_ENABLE_ICUB_ROBOT_ETH` to `ON` again and updtated the repos (`make update-all`).
=> the option change was effective and the`icub-firmware-shared` repo is downloaded.
- rebuilt the superbuild
- run the yarprobotinterface => startup OK

The issue is not reproducible. Unless @S-Dafarra disagrees we can close the issue.

S-Dafarra, 2019-01-21:
Happy to hear that, no objections then!

nunoguedelha, 2019-01-21:
Thx! closing.

nunoguedelha, 2019-01-21:
..er... @traversaro has to do it.

closed 2019-01-21 by traversaro