robotology/robotology-superbuild #131: QApplication file not found

lrapetti, 2018-10-25:
I am trying to install the robotology-superbuild in MacOS, but I am getting the following error message:
```
[ 62%] Built target joystickCheck
[ 62%] Automatic MOC for target iCubGui
[ 62%] Built target iCubGui_autogen
[ 62%] Building CXX object src/tools/iCubGui/src/CMakeFiles/iCubGui.dir/animationview.cpp.o
In file included from /Users/lrapetti/robotology-superbuild/robotology/ICUB/src/tools/iCubGui/src/animationview.cpp:35:
In file included from /Users/lrapetti/robotology-superbuild/robotology/ICUB/src/tools/iCubGui/src/animationview.h:25:
In file included from /Users/lrapetti/robotology-superbuild/robotology/ICUB/src/tools/iCubGui/src/bvh.h:45:
/Users/lrapetti/robotology-superbuild/robotology/ICUB/src/tools/iCubGui/src/bvhnode.h:26:10: fatal error: 'QApplication.h' file not found
#include <QApplication.h>
         ^~~~~~~~~~~~~~~~
1 error generated.
make[5]: *** [src/tools/iCubGui/src/CMakeFiles/iCubGui.dir/animationview.cpp.o] Error 1
make[4]: *** [src/tools/iCubGui/src/CMakeFiles/iCubGui.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [robotology/ICUB/CMakeFiles/YCMStamp/ICUB-build] Error 2
make[1]: *** [CMakeFiles/ICUB.dir/all] Error 2
make: *** [all] Error 2
```

I have installed `Qt` using `homebrew` and `export Qt5_DIR` as explained in the [documentation](https://github.com/robotology/robotology-superbuild#macos).

I also tried to:
- add the global variables as explained during the `hombrew installation` of `qt`:
```
For compilers to find qt you may need to set:
  export LDFLAGS="-L/usr/local/opt/qt/lib"
  export CPPFLAGS="-I/usr/local/opt/qt/include"

For pkg-config to find qt you may need to set:
  export PKG_CONFIG_PATH="/usr/local/opt/qt/lib/pkgconfig"
```
- add all the directories as explained in https://github.com/robotology/robotology-superbuild/issues/4#issuecomment-349363435
- set the directories to `/usr/local/opt/qt5/...` instead of `/usr/local/opt/qt/...` (I see this two folders in `/usr/local/opt/` after the `homebrew` installation)

But I am still getting the same error.

traversaro, 2018-10-25:
Are you using `make` as build tool? Which macOS version are you using? Can you go in the build directory of `icub-main` (`build/robotology/ICUB`) and can you try there to compile by running `make VERBOSE=1`?

traversaro, 2018-10-25:
By the way, that inclusion does not seem to be used at all in that file. Can you try to directly remove that line?

lrapetti, 2018-10-25:
> By the way, that inclusion does not seem to be used at all in that file. Can you try to directly remove that line?

@traversaro 
I removed that line and it compiled without error.

traversaro, 2018-10-25:
Can you prepare a PR against `icub-main` master with this modification? Being a one-line change you can also do it easily from GitHub's web interface. By the way, the fact  that a similar block is completely commented in https://github.com/robotology/icub-main/blob/4f5881acafab693f763dd277f87a8996fdea01a3/src/tools/iCubGui/src/animationview.cpp#L18 makes me wonder if all that include block is useful at all, but this can be verified separately.

lrapetti, 2018-10-25:
PR at https://github.com/robotology/icub-main/pull/558, we can close this issue after the merge.

I have also tried to compile commenting the whole block in `ifdef __APPLE__`:
```
#include <OpenGL/glu.h>
#include <GLUT/glut.h>
#include <QApplication.h>
``` 
and it worked fine. 
It has to be verified if also the block for `WIN32`/`WIN64` can be removed, and then probably the whole block can be removed.

lrapetti, 2018-10-26:
PR was merged https://github.com/robotology/icub-main/commit/a3cb8062ff66b63fd81d6b483a15515e741f9299

closed 2018-10-26 by lrapetti