robotology/robotology-superbuild #134: Qt problems with latest Qt from vcpkg

traversaro, 2018-11-12:
@InesSorrentino experienced several problems related to running the **installed** version of `YARP` and `ICUB` GUIs when using the latest version of Qt installed via `vcpkg` . 

For some reason, the version of the executable contained in the `build` directory seem to be working fine. As there were recently some big changes in how Qt is installed by vcpkg (https://github.com/Microsoft/vcpkg/pull/4164 and https://github.com/Microsoft/vcpkg/pull/4629), it would be worth to check it this problems are actually reproducible.

claudiofantacci, 2018-11-12:
I guess they recompiled everything a was still not working. Am I right?
Speaking about QML and `vcpkg`, there is a specific folder in the vcpkg root folder that is specifically for QML things (not an expert of it, so I'm just throwing whatever comes into my mind here and there). Maybe we should have a look at it and understand whether we have to add folders in `PATH` or something similar.
Anyway, I'm still not 100% sure that the problem is related to `vcpkg` specifically.

InesSorrentino, 2018-11-12:
No, I am not using the vcpkg toolchain.

traversaro, 2018-11-12:
> I guess they recompiled everything a was still not working. Am I right?

Something changed: the executables in the build directory were working correct, but the installed were not working properly: in particular after the line `engine.load(QUrl(QStringLiteral("qrc:/main.qml")))` in https://github.com/robotology/icub-main/blob/4f5881acafab693f763dd277f87a8996fdea01a3/src/tools/iCubSkinGui-qt/src/main.cpp#L27 the `engine.rootObjects()` data structure had size 0, while it typically had size 1 in correctly working applications. Interesting, other applications like `yarpmotorgui` and `yarplogger` were instead working fine also from when launched from the installation directory. Even more interestingly, the `yarpamanager` application was not correctly loading the icons when launched from the installation directory, while everything was working fine (icon were visualized) when launched from the build directory.

>  Anyway, I'm still not 100% sure that the problem is related to vcpkg specifically.

I am trying to reproduce it on a second setup to be sure about it.

> If you are using the vcpkg toolchain, did you disable VCPKG_APPLOCAL_DEPS? This disables copying the dlls in the executable directory

The documented procedure to use `vcpkg` with the `robotology-superbuild` is https://github.com/robotology/robotology-superbuild#vcpkg (TL;DR: no toolchain for us) . The use of the toolchain is not convenient  because it is propagated to subprojects unless you modify the `Build***.cmake` files (that is what we do in some private superbuilds in which we want to do cross-compilation).

traversaro, 2018-11-19:
Another test that it may be worth doing would be to compile YARP using the vcpkg's toolchain, instead of manually setting the enviroment variables. I do not expect the result to change, but it may be worth doing this as an additional check.

DanielePucci, 2018-12-18:
@claudiofantacci any news by chance on the [above test](https://github.com/robotology/robotology-superbuild/issues/134#issuecomment-440185120)?

claudiofantacci, 2018-12-18:
Sorry guys, I froze my `vcpkg` builds to polish up some code.
I'll start the test now and hope to be back to you as soon as possible.

claudiofantacci, 2018-12-19:
> I reproduced the issue experienced by @InesSorrentino (and by @claudia-lat , even if it was not reported here) with vcpkg commit [Microsoft/vcpkg@8dc8d0e](https://github.com/Microsoft/vcpkg/commit/8dc8d0e0c97550a95b764287adbad90b7df7d11d) .
> 
> To make sure that it was actually a vcpkg issue, I tried also to download the latest Qt 5.11.2 binaries from https://www.qt.io/ , and with those everything (yarpmanager with icons, yarpview) is working fine, so the issue seems to be related to vcpkg indeed.
> 
> In particular, the issue seems to be related to the [Qt Resource System](http://doc.qt.io/qt-5/resources.html). `yarpmanager` is not able to load the icons because they are embedded in the binary using the Qt Resource System, and the QML-based application such as `yarpview` do not start because they are unable to load the `"qrc:/main.qml"` file. Ideally, we should try to reproduce the issue in a small example (so-called [Minimal Working Example](https://en.wikipedia.org/wiki/Minimal_working_example) ) and report it to `vcpkg`. In the meanwhile, it may be worth to modify the superbuild readme to mention this problem, and discourage users to use vcpkg (we could also just remove the vcpkg section from the README).

I was able to fix everything, _yet again_, by adding the following `qt.conf` in the YARP bin installation folder:

```ini
[Paths]
Prefix = C:/Users/cfantacci/vcpkg/installed/x64-windows
Documentation = share/qt5/doc
Headers = include
Libraries = lib
Binaries = tools/qt5
LibraryExecutables = tools/qt5
Plugins = plugins
Qml2Imports = qml
Data = share/qt5
ArchData = share/qt5
HostData = share/qt5
HostBinaries = tools/qt5
```

Now everything works as expected.

Unfortunately, I tried to fix this issue on `vcpkg` in the past with https://github.com/Microsoft/vcpkg/pull/3280 that was partially reverted https://github.com/Microsoft/vcpkg/commit/473d63c4edf1ab704b597abc4987372712b005f6.
THis is very unofrtunate. We should really try to find a definitive solution to this.

cc @pattacini @randaz81

traversaro, 2018-12-19:
>  This is very unofrtunate. We should really try to find a definitive solution to this.

I guess the only permanent solution is to have some tests continuously running, either in vcpkg itself or in an external repo: building Qt is extremely slow, but perhaps with the new Azure CI system it should be doable. Related vcpkg issue: https://github.com/Microsoft/vcpkg/issues/72, in particular https://github.com/Microsoft/vcpkg/issues/72#issuecomment-300709680 for the "external" test solution.

claudiofantacci, 2018-12-19:
The problem is that this scenario si very difficult to test, especially if it involves GUIs. You may have a program silently failing, or just not showing up the icons and you would notice that (unless I'm missing something test-wise). This is clearly an issue that must come from users, and possiby a patch from users themselves. I don't find a way out other than discussing with `vcpkg` people about this kind of issues and to understand whether we are misusing something or not.

traversaro, 2018-12-19:
> The problem is that this scenario si very difficult to test, especially if it involves GUIs. You may have a program silently failing, or just not showing up the icons and you would notice that (unless I'm missing something test-wise). 

Are you sure? If it is just a qrc problem, we just need to write a test that tries to access a resource using qrc, and fails if it is not able to find it.

claudiofantacci, 2018-12-19:
> Are you sure? If it is just a qrc problem, we just need to write a test that tries to access a resource using qrc, and fails if it is not able to find it.

You are right. I was focusing my thoughts on the GUI, and not on the Qt Resource System. 👍 

Apart from this, are we going to try discussing with `vcpkg` people that this issue is still inplace? Should we ping about this in the other issue (or pulling people in here) or open an issue in `vcpkg`?

traversaro, 2018-12-19:
If you have a clear understanding of what the issue is (apart from the fact that `qt.conf` fixes it, I personally still do not understand the actual issue), I think we can open an issue `vcpkg` even without a MWE.

claudiofantacci, 2018-12-19:
> If you have a clear understanding of what the issue is (apart from the fact that `qt.conf` fixes it, I personally still do not understand the actual issue), I think we can open an issue `vcpkg` even without a MWE.

I think I found the problem! Let me double check it and report it here.

claudiofantacci, 2018-12-19:
If you open the `Qt5Core.dll` with an hex-editor you will get something like the following list somewhere in the file:
```
qt_prfxpath=C:/Users/cfantacci/vcpkg/installed/x64-windows

...

C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/share/qt5/doc
include
lib
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/share/qt5/bin
bin
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/plugins
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/share/qt5/imports
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/qml
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/share/qt5
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/share/qt5
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/share/qt5/translations
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/share/qt5/examples
tests 
```
Now, by my understanding, every paths that is used by `Qt` start with `qt_prfxpath` and is expanded with the other entries of the list. Not sure if the expantion happens only to relative paths, but I would say so.
If I look at the directory `C:\Users\cfantacci\vcpkg\packages\qt5-base_x64-windows` I have
```
Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----        18-Dec-18     12:49                bin
d-----        18-Dec-18     12:49                debug
d-----        18-Dec-18     12:33                include
d-----        18-Dec-18     12:50                lib
d-----        18-Dec-18     12:50                plugins
d-----        18-Dec-18     12:50                share
d-----        18-Dec-18     12:49                tools
-a----        18-Dec-18     12:50             46 BUILD_INFO
-a----        18-Dec-18     12:50            331 CONTROL
```
for the directory `C:\Users\cfantacci\vcpkg\packages\qt5-base_x64-windows\share`
```
Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----        18-Dec-18     12:49                cmake
d-----        18-Dec-18     12:49                qt5
d-----        18-Dec-18     12:50                qt5-base
```
for the directory `C:\Users\cfantacci\vcpkg\packages\qt5-base_x64-windows\share\qt5`
```
Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----        18-Dec-18     12:32                debug
d-----        18-Dec-18     12:49                doc
d-----        18-Dec-18     12:49                mkspecs
```
and finally for the directory `C:\Users\cfantacci\vcpkg\packages\qt5-base_x64-windows\share\qt5-base`
```
Mode                LastWriteTime         Length Name
----                -------------         ------ ----
-a----        13-Sep-18      6:25           8173 copyright
```
Looking back at our list, the following paths are wrong:
```
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/share/qt5/imports
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/qml
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/share/qt5/translations
C:/Users/cfantacci/vcpkg/packages/qt5-base_x64-windows/share/qt5/examples
```
This paths are overridden by `qt.conf` and the file that I syntesized and reported in the previous points to the correct folder, in particular, I guess, for the `qml` folder. I'm not 100% sure that this is the real cause of the failure, but for sure there are some mistakes that are worth reporting upstream with an issue. What do you think?

traversaro, 2018-12-19:
> What do you think?

I agree.

claudiofantacci, 2019-02-27:
I think we can finally close this issue as the PR https://github.com/Microsoft/vcpkg/pull/5456 in `vcpkg` is fixed.

traversaro, 2019-03-05:
I would prefer if someone successfully tested a complete `robotology-superbuild` install (at least of the `CORE` profile) before closing this issue.

claudiofantacci, 2019-03-10:
Yes, also there has been another [PR](https://github.com/Microsoft/vcpkg/pull/5608) that corrected the tools locations and hopefully did not regress to the other full path issue in the Qt5Core.

I really fear that we are at an empasse with Qt under `vcpkg`.

traversaro, 2019-07-29:
Possibly this could be a related issue: https://github.com/modelica/fmi-standard/pull/586#issuecomment-516024146 .

traversaro, 2019-09-10:
@pattacini you recently updated vcpkg, did you experienced any problem such as the one described in this issue?

pattacini, 2019-09-10:
@traversaro yes the latest qt makes our GUI quite screwed up as in the screenshot below, where SVG icons are not rendered properly:

[image]

traversaro, 2019-09-10:
`yarpmanager` icons indeed look bad, but `yarpview` works fine?

pattacini, 2019-09-11:
@traversaro that's correct; `yarpview` is rendered nicely.

traversaro, 2019-09-11:
Great, so the problem is much less impacting w.r.t. to the original problem for which this issue was opened. In that case, it probably makes sense to reintroduce vcpkg docs in the README.

pattacini, 2019-10-02:
@traversaro I can confirm that the latest Qt version `5.12.5` built via vcpkg solves the SVG icons problem 🎉

traversaro, 2019-10-08:
> @traversaro I can confirm that the latest Qt version 5.12.5 built via vcpkg solves the SVG icons problem tada

For reference: @pattacini in his tests only tested executable in the build directory of YARP, without installing them.

traversaro, 2020-02-18:
This has been tested by several people, so we can finally close it and return to document the use of vcpkg for dependencies.

closed 2020-02-18 by traversaro