robotology/robotology-superbuild #136: Errors related to the inverse kinematics (iDynTree) while compiling the superbuild in Win 32

AngeloRendiniello, 2018-11-19:
I encountered this errors on Visual Studio while trying to build the superbuild with the 32 bit versions of the dipendencies for windows

Error	C2397	conversion from 'unsigned int' to 'ptrdiff_t' requires a narrowing conversion [C:\Users\arendiniello\robotology-superbuild\build\robotology\iDynTree\src\inverse-kinematics\idyntree-inverse-kinematics.vcxproj]	iDynTree (ExternalProjectTargets\iDynTree\iDynTree)	c:\users\arendiniello\robotology-superbuild\robotology\idyntree\src\inverse-kinematics\src\inversekinematicsnlp.cpp	1231	
Error	C2397	conversion from 'unsigned int' to 'ptrdiff_t' requires a narrowing conversion [C:\Users\arendiniello\robotology-superbuild\build\robotology\iDynTree\src\inverse-kinematics\idyntree-inverse-kinematics.vcxproj]	iDynTree (ExternalProjectTargets\iDynTree\iDynTree)	c:\users\arendiniello\robotology-superbuild\robotology\idyntree\src\inverse-kinematics\src\inversekinematicsnlp.cpp	89	
Error	C2397	conversion from 'unsigned int' to 'ptrdiff_t' requires a narrowing conversion [C:\Users\arendiniello\robotology-superbuild\build\robotology\iDynTree\src\inverse-kinematics\idyntree-inverse-kinematics.vcxproj]	iDynTree (ExternalProjectTargets\iDynTree\iDynTree)	c:\users\arendiniello\robotology-superbuild\robotology\idyntree\src\inverse-kinematics\src\inversekinematicsnlp.cpp	303	
Error	C2397	conversion from 'unsigned int' to 'ptrdiff_t' requires a narrowing conversion [C:\Users\arendiniello\robotology-superbuild\build\robotology\iDynTree\src\inverse-kinematics\idyntree-inverse-kinematics.vcxproj]	iDynTree (ExternalProjectTargets\iDynTree\iDynTree)	c:\users\arendiniello\robotology-superbuild\robotology\idyntree\src\inverse-kinematics\src\inversekinematicsnlp.cpp	1209	
Error	C2397	conversion from 'unsigned int' to 'ptrdiff_t' requires a narrowing conversion [C:\Users\arendiniello\robotology-superbuild\build\robotology\iDynTree\src\inverse-kinematics\idyntree-inverse-kinematics.vcxproj]	iDynTree (ExternalProjectTargets\iDynTree\iDynTree)	c:\users\arendiniello\robotology-superbuild\robotology\idyntree\src\inverse-kinematics\src\inversekinematicsnlp.cpp	1239	
Error	C2397	conversion from 'unsigned int' to 'ptrdiff_t' requires a narrowing conversion [C:\Users\arendiniello\robotology-superbuild\build\robotology\iDynTree\src\inverse-kinematics\idyntree-inverse-kinematics.vcxproj]	iDynTree (ExternalProjectTargets\iDynTree\iDynTree)	c:\users\arendiniello\robotology-superbuild\robotology\idyntree\src\inverse-kinematics\src\inversekinematicsnlp.cpp	1307

traversaro, 2018-11-19:
This seems to be a legit compilation error when compiling for 32-bit. Our workaround was to disable the compilation of Inverse Kinematics component by setting `IDYNTREE_USES_IPOPT` to `FALSE`, but this should be properly addressed. Even better, we should have a CI checking that the superbuild compiles correctly on Win32 (i.e. Windows 32-bit).

While support for Windows 32-bit compilation may seems something useless, it is important to note that several proprietary software are just distributed as 32-bit binaries, and some Windows API only work for 32-bit binaries.

traversaro, 2019-01-11:
Fixed by https://github.com/robotology/idyntree/pull/506 .

closed 2019-01-11 by traversaro