robotology/robotology-superbuild #138: Document what the ROBOTOLOGY_USES_OCTAVE option is actually installing

traversaro, 2018-11-22:
In particular, it would be good to specify that it does not install anything unless `ROBOTOLOGY_ENABLE_DYNAMICS` is selected.

traversaro, 2018-11-22:
cc @InesSorrentino

InesSorrentino, 2018-11-23:
There are several problems in `FindOctave.cmake` .

1) It is not able to find the executable octave-config.

```
find_program( OCTAVE_CONFIG_EXECUTABLE
              NAMES octave-config
            )
```

By specifying the path that contains the searched executable, the problem is solved.

```
find_program( OCTAVE_CONFIG_EXECUTABLE
              HINTS "$ENV{SystemDrive}/Octave/Octave-4.4.1/bin"
              NAMES octave-config
            )
```

2) It is not able to execute octave-config -p with the option PREFIX.

```
execute_process ( COMMAND ${OCTAVE_CONFIG_EXECUTABLE} -p PREFIX
                    OUTPUT_VARIABLE OCTAVE_ROOT_DIR
                    OUTPUT_STRIP_TRAILING_WHITESPACE )
```

The problem is that in [this commit](https://bitbucket.org/carandraug/octave/commits/b95c430c06492715c5b318ec068443a9fa7d2df9#chg-src/octave-config.in.cc) of the cmake community, the options has been changed. Indeed if you execute the command `octave-config -h` you can check the new options.
To solve this problem for now I substituted the option `PREFIX` with the option `OCTAVE_EXEC_HOME` in FindOctave.cmake, but I'll open an issue also on cmake to ask information about this.

3) Since Octave is compiled with MinGW and the MinGW libraries are not compatible with VisualStudio, some libraries of Octave can not be found.

```
find_library( OCTAVE_OCTINTERP_LIBRARY
              NAMES octinterp liboctinterp
              HINTS ${OCTAVE_LIBRARIES_PATHS}
            )
find_library( OCTAVE_OCTAVE_LIBRARY
              NAMES octave liboctave
              HINTS ${OCTAVE_LIBRARIES_PATHS}
            )
find_library( OCTAVE_CRUFT_LIBRARY
              NAMES cruft libcruft
              HINTS ${OCTAVE_LIBRARIES_PATHS}
            )
```

traversaro, 2020-04-30:
> In particular, it would be good to specify that it does not install anything unless `ROBOTOLOGY_ENABLE_DYNAMICS` is selected.

This is  not true after yarp-matlab-bindings have been moved to the `Core` profile.

traversaro, 2020-04-30:
> 2. It is not able to execute octave-config -p with the option PREFIX (Windows and Linux). 

This was fixed by https://github.com/robotology/ycm/pull/333 .

> 3. In Windows since Octave is compiled with MinGW and the MinGW libraries are not compatible with VisualStudio, some libraries of Octave can not be found.

Related issue: https://github.com/robotology/robotology-superbuild/issues/139 .