robotology/robotology-superbuild #141: Travis CI on macOS is broken

traversaro, 2018-12-07:
It is the usual problem `brew install package` fails if a package is already installed. In the past @francesco-romano solved it with a script that checked if a package was installed or not, but I think the best option is probably to just add a `Brewfile` to the repo (that we can also use in the macOS installation instructions) and just use that for installing dependencies also in Travis. 

cc @DanielePucci

DanielePucci, 2018-12-07:
Maybe @lrapetti can take upon this?

drdanz, 2018-12-07:
Have you seen the [homebrew travis addon](https://docs.travis-ci.com/user/installing-dependencies/#installing-packages-on-os-x)? I fixed all the issues with travis on macos in YARP by installing the packages with the addon rather than manually.

traversaro, 2018-12-07:
@drdanz Thanks! That is even better.

lrapetti, 2018-12-18:
> Maybe @lrapetti can take upon this?

@traversaro
I may take upon this issue

traversaro, 2018-12-18:
GREAT! Let me know if you need any help. I think that @drdanz pointer is a great starting point, even if I am not sure how it interact with the multi-stage stuff that @diegoferigo originally used for WB-Toolbox and that we copy&pasted around in https://github.com/robotology/robotology-superbuild/pull/108 .

traversaro, 2018-12-19:
Relevant issue for using bundle: https://github.com/Homebrew/homebrew-bundle/issues/313 . @lrapetti

closed 2018-12-21 by traversaro