robotology/robotology-superbuild #143: Add ROBOTOLOGY_USES_OCULUS dependency option

traversaro, 2018-12-11:
As the `LibOVR` library required by YARP for Oculus integration is proprietary, I think it could make sense to have a `ROBOTOLOGY_USES_OCULUS` (or `ROBOTOLOGY_USES_LIBOVR`, but I think that referring to the `OCULUS` is more clear) option similar to the one that we have for Gazebo and scripting based languages such as MATLAB , that are documented in https://github.com/robotology/robotology-superbuild#dependencies-specific-documentation .

This requires either to add `glew` and `glfw3` as packages in the superbuild, or we need to solve https://github.com/robotology-playground/robotology-additional-dependencies/issues/2 .

traversaro, 2018-12-11:
cc @drdanz what you think?

traversaro, 2018-12-14:
I think if the Oculus SDK can be easily download from a fixed URL (it may be impossible directly from https://developer.oculus.com/downloads/package/oculus-sdk-for-windows/ , but I think nobody will complain too much if we setup a mirror with all the legalese) a step that would drastically simplify the process would be to have a `oculusdk-cmake-wrapper` repo that would simpliy: 
*  Download the SDK using CMake's FetchContent 
* Install it in the superbuild install prefix, in a way such that it can be found by the `FindLibOVR` shipped by YCM

Ideally, it would be great to simplify the process of setting up the Oculus machine to simply: 
* Install dependencies 
* Configure and run the superbuild with `ROBOTOLOGY_USES_OculusSDK` set to ON.

drdanz, 2018-12-28:
It's not so easy unfortunately.
The SDK needs to be recompiled after changing `/MD` and `/MDd` to `/MT` and `/MTd`.
I think that the easiest way is to package it for vcpkg

traversaro, 2019-02-02:
Note: 
 * this issue was actually fixed by 
`glew` and `glfw3` have been added to the robotology-additional-dependencies installer. The new installers are available at https://github.com/robotology-playground/robotology-additional-dependencies/releases/tag/v0.2.0 .

closed 2019-02-02 by traversaro