robotology/robotology-superbuild #146: Document usage to WSL to run Gazebo-dependent software on Windows

traversaro, 2018-12-14:
This is common workaround used in the lab, but it is not proper documented so people sometime are not aware of it.

traversaro, 2018-12-31:
After https://bitbucket.org/osrf/gazebo/pull-requests/3009/adding-check-to-make-sure-automatic/diff setting `GAZEBO_IP` it is probably not necessary anymore.

aerydna, 2019-03-07:
long ago i found out that during the configuration phase of the WSL superbuild it happens that cmake find some packages belonging to the other superbuild (the windows one) in mnt/robot-code/etc and it fails. i don't know what kind of witchcraft happens (ther are no variable in the bashrc pointing to those folder) in cmake and the only workaround i found was  to rename the windows folder, configure the wsl superbuild and restore the name on windows.

this happens on both the dedicated oculus rift alienware.

traversaro, 2019-03-07:
> i don't know what kind of cmake witchcraft happens (ther are no variable in the bashrc pointing to those folder) and the only workaround i found was to rename the windows folder, configure the wsl superbuild and restore the name on windows.

We  had the same problem with @AngeloRendiniello . The problem is due to the `PATH` enviroment variable, that is used in `find_package` search strategy (see point 4 of https://cmake.org/cmake/help/v3.10/command/find_package.html). 
Our solution was:
> I had a similar problem related to CMake in WSL finding CMake packages compiled for Windows. 
To remove *all* the path starting with `/mnt/` from the PATH, I used this script in my `.bashrc` (based on https://stackoverflow.com/questions/370047/what-is-the-most-elegant-way-to-remove-a-path-from-the-path-variable-in-bash ): 
~~~
export PATH=`echo ${PATH} | awk -v RS=: -v ORS=: '/mnt/ {next} {print}'`
~~~

traversaro, 2019-03-07:
Another explanation I found: 
> The problem is much more straightforward: all the env variables from Windows are passed to WSL. This includes all the PATH's of dependencies installed by the robotology-installers. `find_package` search path include all the directories contained in `PATH`, with "entries ending in /bin or /sbin are automatically converted to their parent directories" (check point 5 of https://cmake.org/cmake/help/v3.13/command/find_package.html#search-procedure). This means that `find_package` on WSL finds the Windows CMake packages, but clearly with a few exceptions (header only libraryies) the Windows library cannot be linked with the WSL library/binary.

traversaro, 2019-03-07:
By the way @aerydna why do you need to run the robotology-superbuild on  WSL?

drdanz, 2019-03-08:
> The problem is much more straightforward: all the env variables from Windows are passed to WSL. 

This is horrible, I think the way to deal with it is to modify the WSL .bashrc to remove any occurrence of any path outside the WSL box from all the environment variables.

traversaro, 2019-03-08:
I guess the cargo-culted one liner in https://github.com/robotology/robotology-superbuild/issues/146#issuecomment-470732480 can also be used for other env variables.

aerydna, 2019-03-08:
> By the way @aerydna why do you need to run the robotology-superbuild on WSL?
i use gazebo on WSL. and it's very useful to install all the stuff to run icub on gazebo with the superbuild, it saves me a lot of time

aerydna, 2019-03-08:
anyway thanks a lot for the solution

drdanz, 2019-03-08:
> I guess the cargo-culted one liner in https://github.com/robotology/robotology-superbuild/issues/146#issuecomment-470732480 can also be used for other env variables.

This is almost working (The only issue is the final `:`)

```bash
for var in $(env | awk {'FS="="} /\/mnt\//{print $1}'); do echo export ${var}=\"$(echo ${!var} | awk -v RS=: -v ORS=: '/\/mnt\// {next} {print $1}')\" ; done
```

(remove the `echo` in order to use it for real)

traversaro, 2019-03-11:
> i use gazebo on WSL. and it's very useful to install all the stuff to run icub on gazebo with the superbuild, it saves me a lot of time

Have you saw https://github.com/robotology/robotology-superbuild/issues/128#issuecomment-451023509 ?

aerydna, 2019-03-11:
> Have you saw #128 (comment) ?

uh.. no actually! does that mean gazebo on windows (it's not very clear)? i didn't know it was possible. i'll give it a try

traversaro, 2019-03-11:
Basically yes, but the setup/handling of dependencies is still quite complicated (and it is not possible to use the Debug model with the code, see https://github.com/ms-iot/ROSOnWindows/issues/31 .

bemilio, 2019-05-21:
When installing robotology on WSL I was prompted the error 

> this project hasasked CMake to find a package configuration file provided by "Gazebo", butCMake did not find one

Installing libgazebo-dev solved the issue

traversaro, 2019-05-21:
@bemilio  That  should be covered by https://github.com/robotology/robotology-superbuild#system-dependencies-7, but given that you did not find  it, that probably means that it is not too clear in the readme/obscured by the non-linear nature of the README.

closed 2019-09-06 by traversaro