robotology/robotology-superbuild #151: Update the Super-build Windows Installation Environment Variables

kouroshD, 2018-12-19:
I think the Environment varibales for the windows installation should be updated.
These are the necessary variables that a user should add them using Rapid Enviornment Edittor:

- ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX=\<path to code workspace\>\robotology-superbuild\build\install

- ROBOTOLOGY_SUPERBUILD_ROOT=\<path to code workspace\>\robotology-superbuild

- YARP_DATA_DIRS=
  %ROBOTOLOGY_SUPERBUILD_ROOT%\robotology\icub-tests\suits;
  %ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX%\share\ICUBcontrib;
  %ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX%\share\iCub;
  %ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX%\share\yarp;
  \<path to code workspace\>\walking-teleoperation\build\install\share\yarp

- Path=%ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX%\bin

@traversaro  if you agree, please add it the the windows installation guide

Also, I think it is very useful to note to the users about the Dependencies_x64_Release applications, to check for the depenencies of the applications, SDKs, ... if they are correct.
cc @GiulioRomualdi

traversaro, 2018-12-19:
Most of them seems to be already described in https://github.com/robotology/robotology-superbuild#configure-your-environment-2 , or I am missing something? The only missing one is to add ` <path to code workspace>\walking-teleoperation\build\install\share\yarp` to the `YARP_DATA_DIRS`, but unfortunately it does not make sense to do so until the walking-teleoperation repo is added as a package in the superbuild itself. 

> @traversaro if you agree, please add it the the windows installation guide

In general, feel free to directly propose improvements in the form of PRs . For simply documentation fixes or improvements, the easiest option is typically to submit a PR by just modifying the specific document on the GitHub web interface, in this case: https://github.com/robotology/robotology-superbuild/edit/master/README.md .

traversaro, 2019-01-18:
As discussed f2f, we need also to add `CMAKE_PREFIX_PATH` that apparently was never in the instructions. O_o

kouroshD, 2019-01-18:
yes, I will add it.
 Moreover, if you agree, we may suggest the users at the end of the windows installation part, to add the absolute values for the variables, instead of using other variables inside them!

traversaro, 2019-01-18:
> Moreover, if you agree, we may suggest the users at the end of the windows installation part, to add the absolute values for the variables, instead of using other variables inside them!

I totally agree!

traversaro, 2019-02-04:
This was fixed by https://github.com/robotology/robotology-superbuild/pull/155, right?

kouroshD, 2019-02-05:
Yes, they were fixed. so I close the issue.

closed 2019-02-05 by kouroshD