robotology/robotology-superbuild #154: Deprecate ROBOTOLOGY_SUPERBUILD_ROOT env variable?

traversaro, 2018-12-21:
Historically we used the YARP_ROOT, ICUB_ROOT, etc variables are the directories that contain the sources for YARP, iCub, etc. Following this line we did something similar for the defining the `robotology-superbuild`, defining `ROBOTOLOGY_SUPERBUILD_ROOT` in setup.sh and in several setups. 

Since CMake 3.12 the [`CMP0074`](https://cmake.org/cmake/help/git-stage/policy/CMP0074.html) was introduced.
This policy gives a completely different meaning to these variables, and CMake warns when these variables are used.

@drdanz suggested that we switch to a compliant set of variables:

    <Name>_SOURCE_DIR - Source directory
    <Name>_BINARY_DIR - Build directory

I think we can do the same for the robotology-superbuild (even if strictly speaking the superbuild does not contain code, defining `ROBOTOLOGY_SUPERBUILD_SOURCE_DIR` and  `ROBOTOLOGY_SUPERBUILD_BINARY_DIR`. 

Any objection?

@diegoferigo @S-Dafarra @GiulioRomualdi @kouroshD and friends

diegoferigo, 2018-12-22:
Just to have a more clear explanation, with this new policy the `<PackageName>_ROOT` will have the same effect of adding a new install prefix entry to `CMAKE_PREFIX PATH` env var for the package `<PackageName>`?

The users affected are mostly the ones working with the sources on the icub head, I never had the need of using them in my personal setup. I do not have any objections :+1: I always found misleading when I started using CMake the `*_BINARY_*` naming to refer to the build folder, but keeping this consistency is a good idea.

traversaro, 2019-03-13:
> Just to have a more clear explanation, with this new policy the `<PackageName>_ROOT` will have the same effect of adding a new install prefix entry to `CMAKE_PREFIX PATH` env var for the package `<PackageName>`?

Kind-of, I am not sure if the effects are exactly 100% equivalent. 

> The users affected are mostly the ones working with the sources on the icub head, I never had the need of using them in my personal setup. I do not have any objections :+1: I always found misleading [...]

You are right. I think that `<Name>_BINARY_DIR` - Build directory may be a better choice.

drdanz, 2019-03-15:
Sorry, I'm confused by the discussion... My proposal was `BINARY` :sweat_smile:
What is the suggestion? `BINARY` or `BUILD`?

traversaro, 2019-03-15:
Sorry, my fault I modified the original post but that just created confusion, I reverted the changes. For the robotology-superbuild we actually do not have an env variable  for the build, so the problem is not arising, and we are just using `_SOURCE_DIR` in https://github.com/robotology/robotology-superbuild/pull/185 .

traversaro, 2019-03-15:
For other places in which this kind of enviromental variables are used (for example in `.bashrc_iCub` used on the robot, I still think that `_BUILD_DIR` is much more clear for the users w.r.t. `_BINARY_DIR`.

closed 2019-03-17 by traversaro