robotology/robotology-superbuild #161: Should we simplify the setup instructions?

diegoferigo, 2019-01-18:
The README of the superbuild is very (very!) detailed, and it covers a wide variety of configurations. On the other hand, there's so much information that it might create more confusion than benefit, especially to the new users: should I do this or that? In the beginning there were no `setup` scripts, and I can understand that covering all these circumstances was worth.

In my opinion providing a single configuration method to the users would simplify the maintenance of this project, while avoiding many doubts, questions and confusion. Then, if a method does not fit a single use case, it's user responsibility to find a workaround that works specifically for their case. 

If we promote the usage of the scripts, I see benefits for all the user-base since every modification that happens in any of the handled repositories that changes environment variables, is automatically ported to every user setup. Instead, if users manually set env variables, they need to manually edit them every time that something change, possibly leading to arcane errors before they figure it out (not everyone follows announcements).

What I would like to discuss here is if we should simplify the instructions of the superbuild to go towards this direction. I would like especially the feedback of new users of the superbuild. 

Related discussion: https://github.com/robotology/robotology-superbuild/pull/159#discussion_r249003362.

traversaro, 2019-01-18:
> " it covers a wide variety of configurations."/"should I do this or that?"

Can you make an example of this (except for the Matlab case that is already clear in https://github.com/robotology/robotology-superbuild/pull/159#discussion_r249003362)? Which kind of configurations are currently described in the README do you think we can drop. 

> In the beginning there were no setup scripts, and I can understand that covering all these circumstances was worth.

I totally agree on promoting the usage of scripts, but I am not really sure how this is related to the "wide variety of configurations" described in the superbuild. There are some configuration currently described in the README that would that are not supported by the scripts?

Regarding the scripts, I think it could probably make sense to move the documentation of the single environmental variable to update in separate pages (related issue: https://github.com/robotology/robotology-superbuild/issues/46) , while in the main readme only the `setup.*` scripts are advertised. I do not think we should stop documenting the env variables modified by the superbuild, as environmental variables are not something that should taken lightly by users: they are effectively the global state of their computer (even on `*nix`, as ~95% of new users will just drop the `setup.sh` in their `.bashrc` and forget about it). Since I have memory, most of the configuration problems pf the users are due to some environmental variables set to some strange value (classical case: anaconda installation): if they want, users should **be able to know/learn** what the `setup.*` scripts do. 

Related to script support, it is worth noting that we still need to complete https://github.com/robotology/robotology-superbuild/issues/147 . 

> If we promote the usage of the scripts, I see benefits for all the user-base since every modification that happens in any of the handled repositories that changes environment variables, is automatically ported to every user setup.

I am not really concerned about this, to be honest: if a user has a workflow, typically if a new env variable is added is related to some new software/new functionality, and so this is not affecting his workflow, but clearly that is mostly a matter of how the superbuild/single projects are managed: personally I always try to minimize any need of setting environmental variables for their well known problems. 

>  I would like especially the feedback of new users of the superbuild.

I totally agree.

traversaro, 2019-01-19:
I think we agree that the main point is to move the detailed documentation on the env variables to a separate documentation file. Can we rename the issue title to track this?

traversaro, 2019-10-25:
For anyone interested, feel free to open a PR addressing this.

closed 2019-11-26 by traversaro