robotology/robotology-superbuild #163: CMake warning on cmake version 3.13.3 on macOS Mojave

traversaro, 2019-01-22:
~~~sh
CMake Warning (dev) at build/CMakeFiles/ExternalProject.cmake:1570 (math):
  Unexpected character in expression at position 1: "

  Unexpected character in expression at position 6: "

Call Stack (most recent call first):
  build/CMakeFiles/ExternalProject.cmake:2282 (_ep_add_download_command)
  build/CMakeFiles/YCMEPHelper.cmake:1005 (ExternalProject_Add)
  cmake/BuildGazeboYARPPlugins.cmake:34 (ycm_ep_helper)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:235 (include)
  CMakeLists.txt:115 (find_or_build_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
~~~

drdanz, 2019-02-26:
@traversaro is this with the old or the new `ExternalProject` (or both)?

traversaro, 2019-02-26:
I guess old.

traversaro, 2020-04-26:
I don't see the warning on recent CI jobs on macOS, see https://pipelines.actions.githubusercontent.com/ZuyQm6ovz1ad02r7CsxwF1VdDAwrcBAjuVSwcIMiZZdLCwGhMg/_apis/pipelines/1/runs/627/signedlogcontent/6?urlExpires=2020-04-26T09%3A25%3A18.8786700Z&urlSigningMethod=HMACV1&urlSignature=VCplH9p%2BcOkdPyCGVyTzzm4%2FPmuwql0Ns5uEyhmlWqg%3D . 
If the warning emerges again, feel free to open a new issue.

closed 2020-04-26 by traversaro