robotology/robotology-superbuild #169: YARP not building because of missing OPENCV headers

nunoguedelha, 2019-01-29:
### Failure description
The YARP sub-project build fails with the following `Cmake` options:
```
ROBOTOLOGY_ENABLE_CORE
ROBOTOLOGY_ENABLE_DYNAMICS
ROBOTOLOGY_USES_GAZEBO

 YCM_EP_DEVEL_MODE_BlockFactory   OFF                                                                                                 
 YCM_EP_DEVEL_MODE_GazeboYARPPl   ON                                                                                                  
YCM_EP_DEVEL_MODE_ICUB           ON                                                                                                  
 YCM_EP_DEVEL_MODE_ICUBcontrib    OFF                                                                                                 
 YCM_EP_DEVEL_MODE_RTF            ON                                                                                                  
 YCM_EP_DEVEL_MODE_WBToolbox      ON                                                                                                  
 YCM_EP_DEVEL_MODE_YARP           ON                                                                                                  
 YCM_EP_DEVEL_MODE_YCM            OFF                                                                                                 
 YCM_EP_DEVEL_MODE_codyco-modul   ON                                                                                                  
 YCM_EP_DEVEL_MODE_iDynTree       ON                                                                                                  
 YCM_EP_DEVEL_MODE_icub-gazebo    OFF                                                                                                 
 YCM_EP_DEVEL_MODE_icub-gazebo-   ON                                                                                                  
 YCM_EP_DEVEL_MODE_icub-models    OFF                                                                                                 
 YCM_EP_DEVEL_MODE_icub-tests     ON                                                                                                  
 YCM_EP_DEVEL_MODE_whole-body-c   OFF                                                                                                 
 YCM_EP_DEVEL_MODE_wholeBodyInt   OFF                                                                                                 
 YCM_EP_DEVEL_MODE_yarp-matlab-   ON                                                                                                  
 YCM_EP_DEVEL_MODE_yarpWholeBod   OFF                                                                                                 
```
after updating all the repos on the 29/01/2019, and checkout the `devel` branch on all the sub-projects which are in `DEVEL_MODE`.

### Details on the configuration and error message
Error message:
```
CompileC /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YARP/src/yarpdatadumper/YARP.build/Release/yarpdatadumper.build/Objects-normal/x86_64/main.o src/yarpdatadumper/main.cpp normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
    cd /Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP
    export LANG=en_US.US-ASCII
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x c++ -arch x86_64 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -Wno-trigraphs -fpascal-strings -O3 -Wno-missing-field-initializers -Wno-missing-prototypes -Wno-return-type -Wno-non-virtual-dtor -Wno-overloaded-virtual -Wno-exit-time-destructors -Wno-missing-braces -Wparentheses -Wswitch -Wno-unused-function -Wno-unused-label -Wno-unused-parameter -Wno-unused-variable -Wunused-value -Wno-empty-body -Wno-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wno-constant-conversion -Wno-int-conversion -Wno-bool-conversion -Wno-enum-conversion -Wno-float-conversion -Wno-non-literal-null-conversion -Wno-objc-literal-conversion -Wno-shorten-64-to-32 -Wno-newline-eof -Wno-c++11-extensions -DCMAKE_INTDIR=\"Release\" -DBUILDING_YARP -DADD_VIDEO -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -fasm-blocks -fstrict-aliasing -Wdeprecated-declarations -Winvalid-offsetof -mmacosx-version-min=10.12 -Wno-sign-conversion -Wno-infinite-recursion -Wno-move -Wno-comma -Wno-block-capture-autoreleasing -Wno-strict-prototypes -Wno-range-loop-analysis -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YARP/bin/Release/include -I/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/libYARP_OS/include -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YARP/src/libYARP_conf/include -I/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/libYARP_sig/include -isystem /usr/local/include/opencv4 -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YARP/src/yarpdatadumper/YARP.build/Release/yarpdatadumper.build/DerivedSources/x86_64 -I/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YARP/src/yarpdatadumper/YARP.build/Release/yarpdatadumper.build/DerivedSources -Wmost -Wno-four-char-constants -Wno-unknown-pragmas -F/Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YARP/bin/Release -Wall -Wextra -Wsign-compare -Wpointer-arith -Winit-self -Wnon-virtual-dtor -Wcast-align -Wunused -Wvla -Wmissing-include-dirs -Wreorder -Wsizeof-pointer-memaccess -Woverloaded-virtual -Wtautological-undefined-compare -Wmismatched-new-delete -Wparentheses-equality -Wundef -Wredundant-decls -Wunknown-pragmas -Wunused-result -Wc++11-compat -Wheader-guard -Wignored-attributes -Wnewline-eof -Wdangling-else -Wgcc-compat -Wmicrosoft-exists -Wstatic-inline-explicit-instantiation -Wtautological-compare -Winconsistent-missing-override -Wnull-conversion -Wno-unused-parameter -fvisibility=hidden -fvisibility-inlines-hidden -Wdeprecated-declarations -DNDEBUG -fPIE -std=c++11 -MMD -MT dependencies -MF /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YARP/src/yarpdatadumper/YARP.build/Release/yarpdatadumper.build/Objects-normal/x86_64/main.d --serialize-diagnostics /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YARP/src/yarpdatadumper/YARP.build/Release/yarpdatadumper.build/Objects-normal/x86_64/main.dia -c /Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/yarpdatadumper/main.cpp -o /Users/nunoguedelha/dev/robotology-superbuild/build/robotology/YARP/src/yarpdatadumper/YARP.build/Release/yarpdatadumper.build/Objects-normal/x86_64/main.o
/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/yarpdatadumper/main.cpp:428:30: error: use of undeclared identifier 'IplImage'
                int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
                             ^
/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/yarpdatadumper/main.cpp:428:39: error: expected expression
                int frameW=((IplImage*)itemEnd.obj->getPtr())->width;
                                      ^
/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/yarpdatadumper/main.cpp:429:30: error: use of undeclared identifier 'IplImage'
                int frameH=((IplImage*)itemEnd.obj->getPtr())->height;
                             ^
/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/yarpdatadumper/main.cpp:429:39: error: expected expression
                int frameH=((IplImage*)itemEnd.obj->getPtr())->height;
                                      ^
/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/yarpdatadumper/main.cpp:438:52: error: use of undeclared identifier 'CV_FOURCC'
                videoWriter.open(videoFile.c_str(),CV_FOURCC('H','F','Y','U'),
                                                   ^
/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/yarpdatadumper/main.cpp:439:38: error: use of undeclared identifier 'cvSize'
                                 fps,cvSize(frameW,frameH),true);
                                     ^
/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/yarpdatadumper/main.cpp:467:49: error: use of undeclared identifier 'IplImage'
                    cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
                                                ^
/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/yarpdatadumper/main.cpp:467:58: error: expected expression
[... cut ...]
                                                         ^
/Users/nunoguedelha/dev/robotology-superbuild/robotology/YARP/src/yarpdatadumper/main.cpp:467:37: error: no member named 'cvarrToMat' in namespace 'cv'
                    cv::Mat img=cv::cvarrToMat((IplImage*)item.obj->getPtr());
                                ~~~~^
9 errors generated.

** BUILD FAILED **
```

Creating the issue here for now since it could be potentially related to how the superbuild configures the YARP project, and not a YARP intrinsic issue.

nunoguedelha, 2019-01-29:
The definition of `IplImage`struct type and `cvSize` function are missing when building [src/yarpdatadumper/main.cpp](https://github.com/robotology/yarp/blob/master/src/yarpdatadumper/main.cpp). Actually including directly the header `<yarp/sig/IplImage.h>` is now obsolete and the dependency should be included directly from OPEN_CV installed libraries, otherwise we get a warning:
```
YARP_COMPILER_DEPRECATED_WARNING(<yarp/sig/IplImage.h> header is deprecated. Use OpenCV header instead) // Since YARP 3.0.0
```

The required definitions can be found in the Open CV libraries:
```
/usr/local/Cellar/opencv/4.0.1/lib/libopencv_core.4.0.1.dylib
```

that should hold the `IplImage`struct type and `cvSize` function definitions from the header:
```
/usr/local/Cellar/opencv/4.0.1/include/opencv4/opencv2/core/types_c.h
```

In the YARP sub-project CMake configuration (CMakeCache.txt), the option `YARP_USE_OPENCV` is ON, but the `OpenCV_DIR` is set to `/usr/local/lib/cmake/opencv4` which is pointing to `/usr/local/Cellar/opencv/4.0.1/lib/cmake/opencv4` and not `/usr/local/Cellar/opencv/4.0.1/lib` where the libraries are actually stored.

Most probably that should be the problem...

traversaro, 2019-01-29:
Related issue: https://github.com/robotology/yarp/pull/1948 . 
Until YARP is fixed, we should probably document that `opencv@2` should be installed.

nunoguedelha, 2019-01-29:
Actually `/usr/local/lib/` is a mess: it holds individual links to many libraries instead of links to folders like for `qt5`, `perl5`, etc. 
Anyway, although the link to  is missing, where we would find the library `libopencv_core.4.0.1.dylib`, there is the direct link to the library: `/usr/local/lib/libopencv_core.4.0.1.dylib`.

nunoguedelha, 2019-01-29:
After all, `OpenCV_DIR` was set correctly since `/usr/local/Cellar/opencv/4.0.1/lib/cmake/opencv4/OpenCVConfig.cmake` sets the variables:
```
OpenCV_LIBS
OpenCV_INCLUDE_DIRS
OpenCV_LIBRARIES
```

But the include of the OpenCV header was missing anyway in `robotology/YARP/src/yarpdatadumper/main.cpp`. Adding the header
```
#include <opencv2/core/types_c.h>
```
fixed the error on `IplImage`.

nunoguedelha, 2019-01-29:
recaping the missing symbols:
- struct `IplImage` and function `cvSize` -> defined in `<opencv2/core/types_c.h>` -> included in `<opencv2/core/core_c.h>`,
- macro `CV_FOURCC` -> defined in `<opencv2/videoio/legacy/constants_c.h>` -> included in `<opencv2/videoio/videoio_c.h>`
- function `cv::cvarrToMat` -> included in `<opencv2/core/core_c.h>`.

So files to include are:
`<opencv2/core/core_c.h>`
`<opencv2/videoio/videoio_c.h>`

This is just to keep track for me. I had missed your comment:
> Related issue: [robotology/yarp#1948](https://github.com/robotology/yarp/pull/1948) and [robotology/yarp#1672](https://github.com/robotology/yarp/issues/1672) .
> Until YARP is fixed, we should probably document that `opencv@2` should be installed.

Yes, the requirement on `opencv@2` should definitely be documented.

Anyway is OpenCV mostly just for camera features? in that case, I could just deactivate it...

nunoguedelha, 2019-01-29:
> Yes, to be honest `opencv` was never even document in the dependencies of the superbuild in macOS, so I think it is quite a classical choice not to install it.

Honestly, it's the kind of thing I tend to forget if I haven't reinstalled all the stuff for a while, and if it's not documented then the lambda user doesn't know that it should be turned off.
So the best is to document it and eventually add the `opencv@3` to the list of system dependencies. Otherwise it requires additional manual steps: go to YARP sub-project and set the YARP_USES_OPENCV to OFF.
Otherwise CMake could detect the system type (MacOS) and deactivate OpenCV by default.

For the record, I've tested with `opencv@3`:
```
> brew install opencv@3
> brew link --force opencv@3
```
and rebuilt with `YARP_USES_OPENCV` ON. And it builds without issues.

nunoguedelha, 2019-01-29:
I can update later today the Readme.

traversaro, 2019-01-29:
> Honestly, it's the kind of thing I tend to forget if I haven't reinstalled all the stuff for a while, and if it's not documented then the lambda user doesn't know that it should be turned off.
So the best is to document it and eventually add the `opencv@3` to the list of system dependencies. Otherwise it requires additional manual steps: go to YARP sub-project and set the YARP_USES_OPENCV to OFF.

I totally agree, by remark on the missing `opencv` was just on the fact that probably we never put a lot of attention on that dependencies, at least on `macOs`, as we never noticed that the dependency was missing.

lrapetti, 2019-09-18:
> I can update later today the Readme.

The README still needs to be updated

traversaro, 2019-09-18:
As always, contribution from anyone are welcome.

lrapetti, 2019-09-19:
Notice that currently, the brew formula for `opencv@3` is missing `glog` and `ceres-solver` dependencies, that indeed have to be manually installed (see https://github.com/Homebrew/homebrew-core/issues/44415)

lrapetti, 2019-09-24:
> Notice that currently, the brew formula for `opencv@3` is missing `glog` and `ceres-solver` dependencies, that indeed have to be manually installed (see [Homebrew/homebrew-core#44415](https://github.com/Homebrew/homebrew-core/issues/44415))

This has been fixed by https://github.com/Homebrew/homebrew-core/pull/44489

traversaro, 2019-09-24:
Great, I think we can close the issue now?

traversaro, 2019-09-24:
Fixed by https://github.com/robotology/robotology-superbuild/pull/255 .

closed 2019-09-24 by traversaro