robotology/robotology-superbuild #174: Missing projects as make target

lrapetti, 2019-02-04:
Since I update to the latest release of `robotology-superbuild` `master`, it seems I see just few of the project as targets of the superbuild, while the others are not found as target when using `make`, `update-all`, ecc.

The only projects that I can target are:
- `icub-tests`
- `icub-gazebo`
- `icub-gazebo-wholebody`
- `icub-models`
- `whole-body-controllers`

While all the other projects that I can find in `robotology-superbuild/robotology` and `robotology-superbuild/build/robotology` can not be targetes.
_e.g._ 
```
$ cd /robotology-superbuild/build
$ make YARP
make: *** No rule to make target `YARP'.  Stop.
```

I am currently on `MacOS 10.13.6`.

traversaro, 2019-02-04:
Sorry, I though I had replied. I suspect that for some reason the superbuild is finding its own packages: https://github.com/robotology/robotology-superbuild/issues/92 . The packages that you list are the one that do not install a `<package>Config.cmake` file.
Can you post your `CMakeCache.txt` file?

lrapetti, 2019-02-05:
> Can you post your `CMakeCache.txt` file?

```
# This is the CMakeCache file.
# For build in directory: /Users/lrapetti/robotology-superbuild/build
# It was generated by CMake: /usr/local/Cellar/cmake/3.12.3/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Path to a library.
ACE_ACE_LIBRARY_DEBUG:FILEPATH=ACE_ACE_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_ACE_LIBRARY_RELEASE:FILEPATH=/usr/local/lib/libACE.dylib

//Additional compiler flags for ACE
ACE_DEFINITIONS:STRING=

//ACE include directory
ACE_INCLUDE_DIRS:PATH=/usr/local/Cellar/ace/6.5.2/include

//Path to a library.
AVCODEC_LIBRARY_avcodec:FILEPATH=/usr/local/lib/libavcodec.dylib

//Path to a library.
AVDEVICE_LIBRARY_avdevice:FILEPATH=/usr/local/lib/libavdevice.dylib

//Path to a library.
AVFORMAT_LIBRARY_avformat:FILEPATH=/usr/local/lib/libavformat.dylib

//Path to a library.
AVUTIL_LIBRARY_avutil:FILEPATH=/usr/local/lib/libavutil.dylib

//Build the testing tree.
BUILD_TESTING:BOOL=ON

//Path to a program.
BZRCOMMAND:FILEPATH=BZRCOMMAND-NOTFOUND

//The directory containing a CMake configuration file for BlockFactoryCore.
BlockFactoryCore_DIR:PATH=/Users/lrapetti/robotology-superbuild/build/install/lib/cmake/BlockFactoryCore

//The directory containing a CMake configuration file for BlockFactorySimulinkCoder.
BlockFactorySimulinkCoder_DIR:PATH=/Users/lrapetti/robotology-superbuild/build/install/lib/cmake/BlockFactorySimulinkCoder

//The directory containing a CMake configuration file for BlockFactorySimulink.
BlockFactorySimulink_DIR:PATH=/Users/lrapetti/robotology-superbuild/build/install/lib/cmake/BlockFactorySimulink

//The directory containing a CMake configuration file for BlockFactory.
BlockFactory_DIR:PATH=/Users/lrapetti/robotology-superbuild/build/install/lib/cmake/BlockFactory

//Boost atomic library (debug)
Boost_ATOMIC_LIBRARY_DEBUG:FILEPATH=/usr/local/lib/libboost_atomic-mt.dylib

//Boost atomic library (release)
Boost_ATOMIC_LIBRARY_RELEASE:FILEPATH=/usr/local/lib/libboost_atomic-mt.dylib

//Boost chrono library (debug)
Boost_CHRONO_LIBRARY_DEBUG:FILEPATH=/usr/local/lib/libboost_chrono-mt.dylib

[... 2344 lines cut ...]
__pkg_config_arguments_OGRE_PKGC:INTERNAL=OGRE
__pkg_config_arguments_SWSCALE:INTERNAL=libswscale
__pkg_config_arguments_TINYXML2:INTERNAL=tinyxml2
__pkg_config_arguments_UUID:INTERNAL=uuid
__pkg_config_arguments_YAML:INTERNAL=yaml-0.1
__pkg_config_arguments_ZIP:INTERNAL=libzip
__pkg_config_arguments__PC_ACE:INTERNAL=QUIET;ACE
__pkg_config_arguments__PC_SQLITE:INTERNAL=QUIET;sqlite3
__pkg_config_checked_AVCODEC:INTERNAL=1
__pkg_config_checked_AVDEVICE:INTERNAL=1
__pkg_config_checked_AVFORMAT:INTERNAL=1
__pkg_config_checked_AVUTIL:INTERNAL=1
__pkg_config_checked_BULLET:INTERNAL=1
__pkg_config_checked_GTS:INTERNAL=1
__pkg_config_checked_JSONCPP:INTERNAL=1
__pkg_config_checked_OGRE_PKGC:INTERNAL=1
__pkg_config_checked_SWSCALE:INTERNAL=1
__pkg_config_checked_TINYXML2:INTERNAL=1
__pkg_config_checked_UUID:INTERNAL=1
__pkg_config_checked_YAML:INTERNAL=1
__pkg_config_checked_ZIP:INTERNAL=1
__pkg_config_checked__PC_ACE:INTERNAL=1
__pkg_config_checked__PC_SQLITE:INTERNAL=1
prefix_result:INTERNAL=/usr/local/Cellar/libzip/1.5.1/lib
```

traversaro, 2019-02-05:
As you can see from the value of `YARP_DIR`, `ICUB_DIR` , `iDynTree_DIR` etc etc, the superbuild is finding its own packages.
As far as I remember, this should not happen if in your enviroment you just set `CMAKE_PREFIX_PATH`. Can you paste the output of the `env` command?

lrapetti, 2019-02-05:
> Can you paste the output of the `env` command?
```
ROBOTOLOGY_SUPERBUILD_ROOT=/Users/lrapetti/robotology-superbuild
TERM_PROGRAM=Apple_Terminal
MATLABPATH=:/Users/lrapetti/robotology-superbuild/build/install/mex:/Users/lrapetti/robotology-superbuild/build/install/share/WBToolbox:/Users/lrapetti/robotology-superbuild/build/install/share/WBToolbox/images
SHELL=/bin/bash
TERM=xterm-256color
CLICOLOR=1
TMPDIR=/var/folders/68/jjql19w576d27_jsqv87_8hh0000gp/T/
CONDA_SHLVL=1
Apple_PubSub_Socket_Render=/private/tmp/com.apple.launchd.EINfZiMdwb/Render
ROS_MASTER_URI=http://IITICUBLAP079:11311/
GAZEBO_MODEL_PATH=/usr/local/Cellar/gazebo9/9.4.1_1/share/gazebo-9/models:/usr/local/Cellar/gazebo9/9.4.1_1/share/gazebo-9/models::/Users/lrapetti/robotology-superbuild/build/install/share/gazebo/models:/Users/lrapetti/robotology-superbuild/build/install/share/iCub/robots:/Users/lrapetti/robotology-superbuild/build/install/share:/Users/lrapetti/robotology-superbuild/build/install/share/gazebo/models:/Users/lrapetti/robotology-superbuild/build/install/share/iCub/robots:/Users/lrapetti/robotology-superbuild/build/install/share:/Users/lrapetti/robotology-superbuild/robotology/icub-gazebo-wholebody/models:/Users/lrapetti/robotology-playground/human-gazebo
CONDA_PROMPT_MODIFIER=
TERM_PROGRAM_VERSION=404.1
TERM_SESSION_ID=F9705BE6-82B4-4D3C-BFF4-847B27F059CF
USER=lrapetti
CONDA_EXE=/anaconda3/bin/conda
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.ux9ffb1Cmw/Listeners
LSCOLORS=ExFxBxDxCxegedabagacad
GAZEBO_RESOURCE_PATH=/usr/local/Cellar/gazebo9/9.4.1_1/share/gazebo-9:/usr/local/Cellar/gazebo9/9.4.1_1/share/gazebo-9::/Users/lrapetti/robotology-superbuild/build/install/share/gazebo/worlds:/Users/lrapetti/robotology-superbuild/build/install/share/gazebo/worlds:/Users/lrapetti/robotology-superbuild/robotology/icub-gazebo-wholebody/worlds:/Users/lrapetti/robotology-superbuild/robotology/icub-gazebo-wholebody/worlds
PATH=/anaconda3/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/opt/llvm/bin:/Applications/MATLAB_R2018a.app/bin:/Users/lrapetti/robotology-superbuild//build/install/bin:/Users/lrapetti/robotology-superbuild/build/install/bin
CONDA_PREFIX=/anaconda3
PWD=/Users/lrapetti
OGRE_RESOURCE_PATH=/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_6/lib/OGRE
GAZEBO_MASTER_URI=http://localhost:11345
YARP_ROBOT_NAME=icubGazeboSim
XPC_FLAGS=0x0
PS1=\[\033[36m\]\u\[\033[m\]@\[\033[32m\]\h:\[\033[33;1m\]\w\[\033[m\]$ 
YARP_DATA_DIRS=:/Users/lrapetti/robotology-superbuild//build/install/share/yarp:/Users/lrapetti/robotology-superbuild//build/install/share/iCub:/Users/lrapetti/robotology-superbuild//build/install/share/ICUBcontrib:/Users/lrapetti/robotology-superbuild//robotology/icub-tests/suits:/Users/lrapetti/robotology-superbuild/build/install/share/yarp:/Users/lrapetti/robotology-superbuild/build/install/share/iCub:/Users/lrapetti/robotology-superbuild/build/install/share/ICUBcontrib:/Users/lrapetti/robotology-superbuild/robotology/icub-tests/suits:/Users/lrapetti/robotology-superbuild/build/install/share/codyco:/Users/lrapetti/robotology-superbuild/build/install/share/codyco:/Users/lrapetti/loc2/element_torque-control-via-current/device/motorCurrentControl/build/install/share/yarp:/Users/lrapetti/robotology/human-dynamics-estimation/build/install/share/yarp:/Users/lrapetti/robotology/human-dynamics-estimation/app/urdfs
XPC_SERVICE_NAME=0
SHLVL=1
HOME=/Users/lrapetti
GAZEBO_PLUGIN_PATH=/usr/local/Cellar/gazebo9/9.4.1_1/lib/gazebo-9/plugins:/usr/local/Cellar/gazebo9/9.4.1_1/lib/gazebo-9/plugins::/Users/lrapetti/robotology-superbuild/build/install/lib:/Users/lrapetti/robotology-superbuild/build/install/lib
CONDA_PYTHON_EXE=/anaconda3/bin/python
LOGNAME=lrapetti
Qt5_DIR=/usr/local/opt/qt5/lib/cmake/Qt5
LC_CTYPE=UTF-8
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX=/Users/lrapetti/robotology-superbuild/build/install
CONDA_DEFAULT_ENV=base
CMAKE_PREFIX_PATH=:/Users/lrapetti/robotology-superbuild//build/install:/Users/lrapetti/robotology-superbuild/build/install
DISPLAY=/private/tmp/com.apple.launchd.i62yKubha6/org.macosforge.xquartz:0
_=/usr/bin/env
```

traversaro, 2019-02-05:
Mhh, the only strange thing is that the superbuild prefix is specified in the `CMAKE_INSTALL_PREFIX`  twice, I guess due to the `setup.sh` inclusion: 
~~~
CMAKE_PREFIX_PATH=:/Users/lrapetti/robotology-superbuild//build/install:/Users/lrapetti/robotology-superbuild/build/install
~~~
can you try to do a new build, see if the problem persist, and then do a new after removing one path from the `CMAKE_PREFIX_PATH` ?

lrapetti, 2019-02-05:
I tried to do a new build of the superbuild from scratch and the problem, in this case, is not present.

At the same time, I checked the old build after removing one path from `CMAKE_PREFIX_PATH`, and in this case the problem was still there.

traversaro, 2019-02-05:
Ok, so in a new build the problem was not present even if in `CMAKE_PREFIX_PATH` contains two times the `/Users/lrapetti/robotology-superbuild//build/install`?

In the old build, the `<package>_DIR` variables are part of the CMake cache (see https://cgold.readthedocs.io/en/latest/tutorials/variables/cache.html), so regardless of any change of the enviroment once a package is found it will be never "unfound".

traversaro, 2019-02-05:
> And once a package is found, it is no longer compiled by the superbuild? Did I get it right?

Exactly. The rationale behind the YCM's [`find_or_build_package`](http://robotology.github.io/ycm/gh-pages/v0.9/module/FindOrBuildPackage.html) is that the superbuild builds a library only if it is not able to find it (with a suitable version) in the system. The rationale for that is to easily mix libraries that are installed from binaries to reduce the computational burden of the superbuild, but to be honest that feature has not been used a lot lately, and is creating more trouble then benefit. Probably a  `NEVER_USES_SYSTEM_LIBS` option to force every `find_or_build_package` call to be a `build_package` may be indeed really useful.

lrapetti, 2019-02-05:
> Exactly. The rationale behind the YCM's [`find_or_build_package`](http://robotology.github.io/ycm/gh-pages/v0.9/module/FindOrBuildPackage.html) is that the superbuild builds a library only if it is [...]

Ok thanks @traversaro for the explanation.

Anyway, we can say that my specific issue looks solved (things are working), even if we couldn't identify the cause.

traversaro, 2019-02-05:
Ok. Let's close.

lrapetti, 2019-09-19:
This happened to me again, with an installation of the superbuild from scratch after changing some branches (`iDynTree` and `YARP`). Since I know it happened also to others (@kouroshD), I think it may be worth to investigate this further.

traversaro, 2019-09-19:
I was never able to reproduce the problem from a clean system. If you are able to do so, please document the procedure to reproduce this, so we can investigate it.

lrapetti, 2019-09-19:
> I was never able to reproduce the problem from a clean system. If you are able to do so, please document the procedure to reproduce this, so we can investigate it.

I just reproduced it on a clean `robotology-superbuild` in Ubuntu. I have compiled the `core` profile, and then by enabling the `dynamics` flag I no longer see `YARP` target.

traversaro, 2019-09-19:
> I just reproduced it on a clean robotology-superbuild in Ubuntu. I have compiled the core profile, and then by enabling the dynamics flag I no longer see YARP target.

lrapetti, 2019-09-19:
> I guess something is happening in `// .. <something>` ?

I did not do any particular action in the middle, can't be that the problem is in `cmake -DROBOTOLOGY_ENABLE_DYNAMICS:BOOL=ON .` ?

I did not do anything after compiling

> Are you sourcing the `setup.sh`?

No

traversaro, 2019-09-19:
> > I guess something is happening in `// .. <something>` ?
> 
> I did not do any particular action in the middle, can't be that the problem is in `cmake -DROBOTOLOGY_ENABLE_DYNAMICS:BOOL=ON .` ?

I don't know, I created a regression test in https://github.com/robotology/robotology-superbuild/pull/258 that executes exactly those commands, let's see if the issue is reproducible. 
The enviroment is also clean? Or the `.bashrc` already contains something?

lrapetti, 2019-09-19:
> The environment is also clean? Or the `.bashrc` already contains something?

Yes, both in the Ubuntu laptop where I tested it, and the `.bash_profile` in MacOs.

lrapetti, 2019-09-19:
Regression test seems to be passed, I will do again a couple of test and check all the configuration in order to find out if there is any hint to understand when this happens.

isorrentino, 2019-10-01:
I am experiencing the same problem in my laptop.

traversaro, 2019-10-01:
@lrapetti I noticed that on @isorrentino laptop, ROS was installed and sourced. You were able to reproduce the issue in a system without ROS?

lrapetti, 2019-10-01:
> @lrapetti I noticed that on @isorrentino laptop, ROS was installed and sourced. You were able to reproduce the issue in a system without ROS?

I think the first times I experienced it on the new laptop ROS was not yet installed

traversaro, 2019-10-15:
The attempt of reproducing the the issue #174 failed.

traversaro, 2019-10-15:
@lrapetti I am unable to reproduce the issue, if you can even in random laptop, please ping me, thanks!

traversaro, 2019-12-09:
@lrapetti I just realized something: do you remember if when you experienced this issue, you used an external YCM?

lrapetti, 2019-12-10:
> @lrapetti I just realized something: do you remember if when you experienced this issue, you used an external YCM instead of the usual bootstrapped one?

I have always installed it together with the superbuild. Maybe I had it on my system from a previous superbuild.

traversaro, 2020-04-26:
A  definite fix for this issue is provided in https://github.com/robotology/robotology-superbuild/issues/174 . It requires a feature present in a recent YCM, so it will only work only if you select `Unstable` as `ROBOTOLOGY_PROJECT_TAGS`, but as soon as the new feature release of YCM is released, it will also work for `Stable`, and we will be able to update the README and  close this issue.

lrapetti, 2020-10-06:
cc @CarlottaSartore

traversaro, 2020-10-06:
This happened again?

CarlottaSartore, 2020-10-06:
It is happening to me but I do not have the fix in the robotology superbuild yet (#389 ).  I am going to upgrade and check

traversaro, 2020-10-06:
macOS or Ubuntu? If Ubuntu, which version?

CarlottaSartore, 2020-10-06:
I am on Ubuntu 
```
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
```

robotology superbuild commit  e794398d4f5315604ebb68f40384d47d37f6851c

traversaro, 2020-11-27:
As we switched to use YCM 0.12 in https://github.com/robotology/robotology-superbuild/pull/499, now both `Stable` and `Unstable` branches of the robotology-superbuild **rolling update** and the release `v2020.11` that is going to be released next week should not be affected anymore by this. @GiulioRomualdi reported that this is still happening on the iCubGenova04 setup, but that is using an older version of the superbuild. For this reason, I think we can close the issue. If you are experiencing again this problem in a version after v2020.11 or in a latest version, feel free to open a new issue, thanks.

closed 2020-11-27 by traversaro